xnvme/xnvme #1: No targets error while make project after make

SuhoSon, 2020-09-01:
I clone xNVMe repository and since I did not install meson, make is failing.
After install meson, I do re-make the project, but error is occurred like below.
I think it caused by failing to make third-party, but I'm not sure.

``` bash
CMake Error at CMakeLists.txt:255 (message):
  Failed building SPDK/DPDK

-- Configuring incomplete, errors occurred!
See also "/home/oslab4/xNVMe/build/CMakeFiles/CMakeOutput.log".
See also "/home/oslab4/xNVMe/build/CMakeFiles/CMakeError.log".

Configuration failed

Makefile:53: recipe for target 'config' failed
make[1]: *** [config] Error 1
make[1]: Leaving directory '/home/oslab4/xNVMe'
Makefile:47: recipe for target 'default' failed
make: *** [default] Error 2
root@oslab4 ~/xNVMe (master)$
root@oslab4 ~/xNVMe (master)$ make                                                  <-- my command
## xNVMe: make info
OSTYPE:
PLATFORM: Linux
CC: cc
CXX: g++
MAKE: make
CTAGS: ctags
NPROC: 32
## xNVMe: make tags
/bin/sh: 1: ctags: not found
## xNVMe: make default
$( case $( uname -s ) in ( Linux ) echo "make" ;; ( FreeBSD | OpenBSD | NetBSD ) echo "gmake" ;; ( * ) echo Unrecognized ;; esac) build
make[1]: Entering directory '/home/oslab4/xNVMe'
## xNVMe: make info
OSTYPE:
PLATFORM: Linux
CC: cc
CXX: g++
MAKE: make
CTAGS: ctags
NPROC: 32
## xNVMe: make build
cd build && $( case $( uname -s ) in ( Linux ) echo "make" ;; ( FreeBSD | OpenBSD | NetBSD ) echo "gmake" ;; ( * ) echo Unrecognized ;; esac)
make[2]: Entering directory '/home/oslab4/xNVMe/build'
make[2]: *** No targets specified and no makefile found.  Stop.
make[2]: Leaving directory '/home/oslab4/xNVMe/build'
Makefile:74: recipe for target 'build' failed
make[1]: *** [build] Error 2
make[1]: Leaving directory '/home/oslab4/xNVMe'
Makefile:47: recipe for target 'default' failed
make: *** [default] Error 2
```

safl, 2020-09-01:
Hi @SuhoSon,

Thanks for reporting this, it is because it did a partial configuration, that is, it created the build directory but did not produce the cmake output. You should be able to just do:

``
make clean
``

And then do:

``
make
``

Again.

SuhoSon, 2020-09-01:
Yes, I was able to move forward. Thank you @safl !

safl, 2020-09-01:
Great, you are welcome @SuhoSon !

closed 2020-09-01 by safl