xnvme/xnvme #5: Unable to use spdk backend using latest libxnvme version 21

surajk2, 2020-12-15:
Compiling and linking your code with xNVMe
Referring -> https://xnvme.io/docs/latest/backends/xnvme_be_spdk/index.html

**CMakeLists.txt**
target_compile_options(${PROJECT_NAME} PUBLIC -MMD)
target_compile_options(${PROJECT_NAME} PUBLIC -MP)
target_compile_options(${PROJECT_NAME} PUBLIC -MF)
target_compile_options(${PROJECT_NAME} PUBLIC -fPIE)

target_link_libraries(${PROJECT_NAME} -Wl,--whole-archive)
target_link_libraries(${PROJECT_NAME} -Wl,--no-as-needed)
target_link_libraries(${PROJECT_NAME} -lxnvme)
target_link_libraries(${PROJECT_NAME} -Wl,--no-whole-archive)
target_link_libraries(${PROJECT_NAME} -Wl,--as-needed)

**Error:**
[2020-12-15 15:19:58.023298] rpc.c: 216:spdk_rpc_register_method: *ERROR*: duplicate RPC rpc_get_methods registered...
[2020-12-15 15:19:58.023413] rpc.c: 216:spdk_rpc_register_method: *ERROR*: duplicate RPC spdk_get_version registered...
[2020-12-15 15:19:58.023429] rpc.c: 216:spdk_rpc_register_method: *ERROR*: duplicate RPC sock_impl_get_options registered...
[2020-12-15 15:19:58.023439] rpc.c: 216:spdk_rpc_register_method: *ERROR*: duplicate RPC sock_impl_set_options registered...
EAL: RTE_MEMPOOL tailq is already registered
PANIC in tailqinitfn_rte_mempool_tailq():
Cannot initialize tailq: RTE_MEMPOOL

Please suggest something that i can try to make this work.

safl, 2020-12-15:
I can't tell how you are building / linking your app, can you try building it directly with ``gcc``, like so:

```
gcc your_app.c \
	-Wl,--whole-archive -Wl,--no-as-needed \
	-lxnvme \
	-Wl,--no-whole-archive -Wl,--as-needed \
	-laio -luuid -lnuma -pthread \
	-o your_app
```

Thanks!

surajk2, 2020-12-15:
Ok, will try and update

surajk2, 2020-12-15:
Using a simple test app. I just tried using gcc and there is no problem in compiling and linking. Able to get output too.
But not sure why linking twice in my project case. Will debug further and check. 

Result:
#include <stdio.h>

#include <libxnvme.h>

int main(int argc, char **argv)

{

        struct xnvme_dev *dev = xnvme_dev_open("pci:0000:0b:00.0?nsid=1");

        if (!dev) {

                perror("xnvme_dev_open");

                return 1;

        }

        xnvme_dev_pr(dev, XNVME_PR_DEF);

        xnvme_dev_close(dev);

        return 0;

}

[root@localhost test_libxnvme]# gcc test_app.c -Wl,--whole-archive -Wl,--no-as-needed -lxnvme -Wl,--no-whole-archive -Wl,--as-needed -laio -luuid -lnuma -pthread
[root@localhost test_libxnvme]# ./a.out 
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_state_init-936: !xnvme_ident_opt_to_val(opt:cmb_sqs)
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_state_init-939: !xnvme_ident_opt_to_val(opt:css)
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_state_init-945: INFO: dev->nsid: 1, state->cmb_sqs: 0, state->css: 0
Starting SPDK v20.07 / DPDK 20.05.0 initialization...
[ DPDK EAL parameters: spdk --no-shconf -c 0x1 --log-level=lib.eal:6 --log-level=lib.cryptodev:5 --log-level=user1:6 --iova-mode=pa --base-virtaddr=0x200000000000 --match-allocations --file-prefix=spdk_pid7528 ]
EAL: No available hugepages reported in hugepages-1048576kB
EAL: VFIO support initialized
EAL: No legacy callbacks, legacy socket not created
# DBG:xnvme_be_spdk.c:_spdk_env_init-253: INFO: spdk_env_is_initialized: 1
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_state_init-998: INFO: trtype: PCIe, # 1 of 4
EAL:   using IOMMU type 1 (Type 1)
trid:
  trstring: 'PCIE'
  trtype: 0x100
  adrfam: 0x0
  traddr: '0000:0b:00.0'
  trsvcid: ''
  subnqn: ''
  priority: 0x0
# DBG:xnvme_be_spdk.c:attach_cb-517: INFO: nsid: 1
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_state_init-998: INFO: trtype: TCP, # 2 of 4
[2020-12-15 17:16:55.766415] nvme_tcp.c:1620:nvme_tcp_ctrlr_connect_qpair: *ERROR*: Invalid port: 0
[2020-12-15 17:16:55.766465] nvme_tcp.c:1731:nvme_tcp_ctrlr_construct: *ERROR*: failed to connect admin qpair
[2020-12-15 17:16:55.766483] nvme.c: 630:nvme_probe_internal: *ERROR*: NVMe ctrlr scan failed
[2020-12-15 17:16:55.766498] nvme.c: 707:spdk_nvme_probe: *ERROR*: Create probe context failed
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_state_init-1008: FAILED: probe a:1, e:-1, i:0
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_state_init-998: INFO: trtype: RDMA, # 3 of 4
[2020-12-15 17:16:55.766537] nvme.c: 622:nvme_probe_internal: *ERROR*: NVMe trtype 1 not available
[2020-12-15 17:16:55.766553] nvme.c: 707:spdk_nvme_probe: *ERROR*: Create probe context failed
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_state_init-1008: FAILED: probe a:1, e:-1, i:0
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_state_init-998: INFO: trtype: FC, # 4 of 4
# DBG:xnvme_be_spdk.c:_xnvme_be_spdk_ident_to_trid-322: FAILED: unsupported trtype: FC
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_state_init-1001: SKIP/FAILED: ident_to_trid()
[2020-12-15 17:16:55.767007] nvme_qpair.c: 250:nvme_admin_qpair_print_command: *NOTICE*: IDENTIFY (06) qid:0 cid:23 nsid:0 cdw10:00000006 cdw11:02000000 PRP1 0x6be855000 PRP2 0x0
[2020-12-15 17:16:55.767036] nvme_qpair.c: 454:spdk_nvme_print_completion: *NOTICE*: INVALID FIELD (00/02) qid:0 cid:23 cdw0:0 sqhd:0013 p:1 m:0 dnr:0
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_sync_cmd_admin-1341: FAILED: xnvme_req_cpl_status()
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_dev_idfy-860: INFO: !id-ctrlr-zns
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_dev_idfy-886: INFO: failed idfy with csi(ZNS)
[2020-12-15 17:16:55.768522] nvme_qpair.c: 250:nvme_admin_qpair_print_command: *NOTICE*: IDENTIFY (06) qid:0 cid:23 nsid:1 cdw10:00000005 cdw11:00000000 PRP1 0x6be853000 PRP2 0x0
[2020-12-15 17:16:55.768548] nvme_qpair.c: 454:spdk_nvme_print_completion: *NOTICE*: INVALID FIELD (00/02) qid:0 cid:23 cdw0:0 sqhd:0015 p:1 m:0 dnr:0
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_sync_cmd_admin-1341: FAILED: xnvme_req_cpl_status()
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_dev_idfy-894: INFO: not csi-specific id-NVM
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_dev_idfy-895: INFO: falling back to NVM assumption
xnvme_dev:
  xnvme_ident:
    trgt: '0000:0b:00.0'
    schm: 'pci'
    opts: '?nsid=1'
    uri: 'pci:0000:0b:00.0?nsid=1'
  xnvme_be:
    async: {id: 'nvme_driver', enabled: 1}
    sync: {id: 'nvme_driver', enabled: 1}
    attr: {name: 'spdk', enabled: 1}
  xnvme_cmd_opts:
    mask: '00000000000000000000000000000001'
    iomd: 'SYNC'
    payload_data: 'DRV'
    payload_meta: 'DRV'
    csi: 0x0
    nsid: 0x1
    ssw: 12
  xnvme_geo:
    type: XNVME_GEO_CONVENTIONAL
    npugrp: 1
    npunit: 1
    nzone: 1
    nsect: 195353046
    nbytes: 4096
    nbytes_oob: 0
    tbytes: 800166076416
    mdts_nbytes: 1048576
    lba_nbytes: 4096
    lba_extended: 0
# DBG:xnvme_be_spdk.c:_cref_deref-114: INFO: refcount: 0 => detaching

closed 2020-12-15 by surajk2