xnvme/xnvme #6: Unable to open device with PCI URI

pcwilcox, 2020-12-16:
After configuring and installing

```
$  ./configure --enable-debug --disable-tests --disable-examples --disable-tools --disable-tools-fio --enable-be-spdk
$ make
$ sudo make install
```

And building the example application, using the pci identifier to enable the SPDK backend:

```
#include <stdio.h>
#include <libxnvme.h>

int
main(int argc, char** argv)
{
    struct xnvme_dev* dev = xnvme_dev_open("pci://0000:af:00.0");
    if (!dev) {
        perror("xnvme_dev_open");
        return 1;
    }
    xnvme_dev_pr(dev, XNVME_PR_DEF);
    xnvme_dev_close(dev);

    return 0;
}
```

This address corresponds to the NVMe SSD on my system:
```
af:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM173X
```

Expected behavior is something like the example output in the docs. Instead I get:
```
xnvme_dev_open: No such device or address
```

I've tried numerous variations on the PCI string but none have worked. What is the needed configuration to enable the SPDK backend?

safl, 2020-12-16:
Ah, it probably needs the namespace identifier e.g.

```
struct xnvme_dev* dev = xnvme_dev_open("pci://0000:af:00.0?nsid=1");
```

For reference, running the cli-tool:

```
xnvme enum
```

Should output a list of available devices (controllers/namespaces), such as:

```
# xnvme_enumerate()
xnvme_enumeration:
  capacity: 98
  nentries: 2
  entries:
  - {trgt: '0000:03:00.0', schm: 'pci', opts: '?nsid=1', uri: 'pci:0000:03:00.0?nsid=1'}
  - {trgt: '0000:03:00.0', schm: 'pci', opts: '?nsid=2', uri: 'pci:0000:03:00.0?nsid=2'}
```

pcwilcox, 2020-12-17:
This command doesn't seem to be working on my system. I reconfigured without `--disable-tools` and `--disable-tools-fio` just in case, then `make` and `sudo make install` again, and run the CLI tool:

```
$ xnvme enum
# xnvme_enumerate()
xnvme_enumeration:
  capacity: 100
  nentries: 0
  entries: ~
```

safl, 2020-12-17:
Hmm, this is probably due to your NVMe device(s) sharing IOMMU group with devices which cannot unbind from the kernel.

Can you try this:

```
make clean
make config-debug
make
sudo make install
```

And then running:

```
# This will detach your NVMe devices from Kernel for the SPDK/NVMe driver to pick them up
$ sudo xnvme-driver

# This should now output a bunch of debug info, describing what goes wrong
$ sudo xnvme enum
```

If you dump the output from the above then we should be able to find the answer there.

pcwilcox, 2020-12-17:
This seems like progress.

```
$ sudo xnvme-driver
0000:3b:00.0 (8086 2700): Active mountpoints on /dev/nvme0n1, so not binding PCI dev
0000:af:00.0 (144d a824): nvme -> uio_pci_generic
$ sudo xnvme enum
# xnvme_enumerate()
Starting SPDK v20.07 git sha1 1a527e501 / DPDK 20.05.0 initialization...
[ DPDK EAL parameters: xnvme -c 0x1 --log-level=lib.eal:6 --log-level=lib.cryptodev:5 --log-level=user1:6 --iova-mode=pa --base-virtaddr=0x200000000000 --match-allocations --file-prefix=spdk0 --proc-type=auto ]
EAL: No available hugepages reported in hugepages-1048576kB
EAL: No legacy callbacks, legacy socket not created
# DBG:xnvme_be_spdk.c:_spdk_env_init-252: INFO: spdk_env_is_initialized: 1
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-741: INFO: trtype: PCIe, # 1 of 4
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 2
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 3
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 4
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 5
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 6
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 7
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 8
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 9
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 10
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 11
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 12
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 13
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 14
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 15
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 16
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 17
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 18
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 19
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 20
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 21
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 22
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 23
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 24
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 25
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 26
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 27
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 28
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 29
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 30
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 31
# DBG:xnvme_be_spdk.c:enumerate_attach_cb-643: FAILED: skipping inactive nsid: 32
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-741: INFO: trtype: TCP, # 2 of 4
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-753: SKIP: !sys_uri
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-741: INFO: trtype: RDMA, # 3 of 4
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-753: SKIP: !sys_uri
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-741: INFO: trtype: FC, # 4 of 4
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-775: SKIP: enum. of trtype: FC not implemented
# DBG:xnvme_be_linux.c:xnvme_be_linux_state_init-207: state->poll_io: 0
# DBG:xnvme_be_linux.c:xnvme_be_linux_state_init-208: state->poll_sq: 0
# DBG:xnvme_be_linux.c:xnvme_be_linux_state_init-230: id: nvme_ioctl, enabled: 1
# DBG:xnvme_be_linux.c:xnvme_be_linux_state_init-235: got: nvme_ioctl
# DBG:xnvme_be_linux.c:xnvme_be_linux_state_init-259: got: thr
# DBG:xnvme_be_linux.c:xnvme_be_linux_dev_idfy-309: INFO: dev->nsid: 1
# DBG:xnvme_be_linux_nvme.c:ioctl_wrap-84: FAILED: ioctl(NVME_IOCTL_ADMIN_CMD), err(16395), errno(2)
# DBG:xnvme_be_linux_nvme.c:ioctl_wrap-100: INFO: overwr. cpl.status.val(0x0) with '16395'
# DBG:xnvme_be_linux_nvme.c:xnvme_be_linux_nvme_cmd_admin-155: FAILED: ioctl_wrap() err: 16395
# DBG:xnvme_be_linux.c:xnvme_be_linux_dev_idfy-359: INFO: !id-ctrlr-zns
# DBG:xnvme_be_linux.c:xnvme_be_linux_dev_idfy-385: INFO: failed idfy with csi(ZNS)
# DBG:xnvme_be_linux.c:xnvme_be_linux_state_init-207: state->poll_io: 0
# DBG:xnvme_be_linux.c:xnvme_be_linux_state_init-208: state->poll_sq: 0
# DBG:xnvme_be_linux.c:xnvme_be_linux_state_init-230: id: nvme_ioctl, enabled: 1
# DBG:xnvme_be_linux.c:xnvme_be_linux_state_init-230: id: block_ioctl, enabled: 1
# DBG:xnvme_be_linux.c:xnvme_be_linux_state_init-235: got: block_ioctl
# DBG:xnvme_be_linux.c:xnvme_be_linux_state_init-259: got: thr
# DBG:xnvme_be_linux_block.c:_idfy_ctrlr-374: max_hw_sectors_kb: 128
# DBG:xnvme_be_linux.c:xnvme_be_linux_dev_idfy-359: INFO: !id-ctrlr-zns
[... 64 lines cut ...]
    attr: {name: 'spdk', enabled: 1}
  xnvme_cmd_opts:
    mask: '00000000000000000000000000000001'
    iomd: 'SYNC'
    payload_data: 'DRV'
    payload_meta: 'DRV'
    csi: 0x0
    nsid: 0x1
    ssw: 9
  xnvme_geo:
    type: XNVME_GEO_CONVENTIONAL
    npugrp: 1
    npunit: 1
    nzone: 1
    nsect: 6251233968
    nbytes: 512
    nbytes_oob: 0
    tbytes: 3200631791616
    mdts_nbytes: 131072
    lba_nbytes: 512
    lba_extended: 0
# DBG:xnvme_be_spdk.c:_cref_deref-113: INFO: refcount: 0 => detaching
```

This certainly seems like the expected behavior except for the debug messages.

pcwilcox, 2020-12-17:
Okay, I think that has fixed it. I added `intel_iommu=on` to my kernel command line, rebooted, recompiled xNVMe, ran `sudo HUGEMEM=8192 xnvme-driver`, and now the application works:

```
$ sudo hello
xnvme_dev:
  xnvme_ident:
    trgt: '0000:af:00.0'
    schm: 'pci'
    opts: '?nsid=1'
    uri: 'pci:0000:af:00.0?nsid=1'
  xnvme_be:
    async: {id: 'nvme_driver', enabled: 1}
    sync: {id: 'nvme_driver', enabled: 1}
    attr: {name: 'spdk', enabled: 1}
  xnvme_cmd_opts:
    mask: '00000000000000000000000000000001'
    iomd: 'SYNC'
    payload_data: 'DRV'
    payload_meta: 'DRV'
    csi: 0x0
    nsid: 0x1
    ssw: 9
  xnvme_geo:
    type: XNVME_GEO_CONVENTIONAL
    npugrp: 1
    npunit: 1
    nzone: 1
    nsect: 6251233968
    nbytes: 512
    nbytes_oob: 0
    tbytes: 3200631791616
    mdts_nbytes: 131072
    lba_nbytes: 512
    lba_extended: 0
```

Thanks!

closed 2020-12-17 by pcwilcox