xnvme/xnvme #7: xnvme enum is not working

surajk2, 2020-12-17:
[root@localhost xnvme-0.0.21]# xnvme-driver reset
0000:01:00.0 (144d a808): vfio-pci -> nvme
[root@localhost xnvme-0.0.21]# 
[root@localhost xnvme-0.0.21]# xnvme-driver
0000:01:00.0 (144d a808): nvme -> vfio-pci
[root@localhost xnvme-0.0.21]# xnvme enum
# xnvme_enumerate()
Starting SPDK v20.07 / DPDK 20.05.0 initialization...
[ DPDK EAL parameters: xnvme -c 0x1 --log-level=lib.eal:6 --log-level=lib.cryptodev:5 --log-level=user1:6 --iova-mode=pa --base-virtaddr=0x200000000000 --match-allocations --file-prefix=spdk0 --proc-type=auto ]
EAL: No available hugepages reported in hugepages-1048576kB
EAL: VFIO support initialized
EAL: Cannot allocate memzone list
EAL: FATAL: Cannot init memzone
EAL: Cannot init memzone
Failed to initialize DPDK
# DBG:xnvme_be_spdk.c:_spdk_env_init-244: FAILED: spdk_env_init(), err: -19
# DBG:xnvme_be_spdk.c:_spdk_env_init-253: INFO: spdk_env_is_initialized: 0
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-717: FAILED: _spdk_env_init()
# DBG:xnvme_be.c:xnvme_enumerate-716: FAILED: spdk->enumerate(...), err: 'No such process', i: 0
# DBG:xnvme_be_nosys.c:xnvme_be_nosys_enumerate-141: FAILED: not implemented(possibly intentional)
# DBG:xnvme_be.c:xnvme_enumerate-716: FAILED: fbsd->enumerate(...), err: 'Function not implemented', i: 2
xnvme_enumeration:
  capacity: 100
  nentries: 0
  entries: ~

safl, 2020-12-17:
Hi @surajk2,

It is failing because it cannot initialize SPDK, SPDK in turn fails because it cannot initialize DPDK. Which fails to initialize due to the error:

```
EAL: Cannot allocate memzone list
EAL: FATAL: Cannot init memzone
EAL: Cannot init memzone
```

A couple of things to check:

* Are you running multiple instances of your program?
* Could it be that previous execution of your program is still running e.g. a defunct thread, or some other unclean terminated process?

If this is not the case, then it might be the configuration of HUGEMEM on your system. If you haven't already done so, then try:

https://xnvme.io/docs/latest/backends/xnvme_be_spdk/index.html#enabling-vfio-without-limits

Also, when running ``xnvme-driver`` then you can control the HUGEMEM allocation, e.g.
```
HUGEMEM=8192 xnvme-driver
```

surajk2, 2020-12-18:
Restarted PC and tried again. Now getting different error. can you suggest something to fix this issue.

[root@localhost ~]# xnvme enum
# xnvme_enumerate()
Starting SPDK v20.07 / DPDK 20.05.0 initialization...
[ DPDK EAL parameters: xnvme -c 0x1 --log-level=lib.eal:6 --log-level=lib.cryptodev:5 --log-level=user1:6 --iova-mode=pa --base-virtaddr=0x200000000000 --match-allocations --file-prefix=spdk0 --proc-type=auto ]
EAL: No available hugepages reported in hugepages-1048576kB
EAL: VFIO support initialized
EAL: No legacy callbacks, legacy socket not created
# DBG:xnvme_be_spdk.c:_spdk_env_init-253: INFO: spdk_env_is_initialized: 1
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-743: INFO: trtype: PCIe, # 1 of 4
EAL:   0000:01:00.0 VFIO group is not viable! Not all devices in IOMMU group bound to VFIO or unbound
EAL: Requested device 0000:01:00.0 cannot be used
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-743: INFO: trtype: TCP, # 2 of 4
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-753: SKIP: !sys_uri
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-743: INFO: trtype: RDMA, # 3 of 4
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-753: SKIP: !sys_uri
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-743: INFO: trtype: FC, # 4 of 4
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-776: SKIP: enum. of trtype: FC not implemented
# DBG:xnvme_be_nosys.c:xnvme_be_nosys_enumerate-141: FAILED: not implemented(possibly intentional)
# DBG:xnvme_be.c:xnvme_enumerate-716: FAILED: fbsd->enumerate(...), err: 'Function not implemented', i: 2
xnvme_enumeration:
  capacity: 100
  nentries: 0
  entries: ~
[root@localhost ~]#

surajk2, 2020-12-18:
**EAL: 0000:01:00.0 VFIO group is not viable! Not all devices in IOMMU group bound to VFIO or unbound
EAL: Requested device 0000:01:00.0 cannot be used**

safl, 2020-12-18:
Hi @surajk2,
Ok, this is a bit challenging, looks like other devices, could be NICs, GPUs or root complexes which are sharing the IOMMU group with the NVMe device(s). Sometimes it is possible to manually unbind the kernel drivers from the auxiliary devices, if they aren't used.
I have an AMD system in the lab which exhibits the same behaviour, I will take a look and see if I can find a work-around for it.
If possible, can you share system info, e.g. lshw / lspci output?

safl, 2020-12-18:
Another thing to try is to have the device bind with ``uio`` instead of ``vfio``.

To try this, then disable the iommu by adding ``iommu=0`` to your kernel command-line, update your bootloader and reboot.
Then when running ``xnvme-driver`` you will see:

nvme -> uio_pci_generic

Instead of:

nvme -> vfio-pci

safl, 2020-12-27:
Hi @surajk2 ,
Did you have chance to change device attachment or switch to ``uio_pci_generic``?
Best,
Simon

surajk2, 2020-12-29:
Hi @safl 
I followed the steps as you mentioned above. I get the following debug info. Please check.

0000:01:00.0 (144d a808): no driver -> uio_pci_generic
[root@localhost ~]# xnvme enum
; xnvme_enumerate()
Starting SPDK v20.07 / DPDK 20.05.0 initialization...
[ DPDK EAL parameters: xnvme -c 0x1 --log-level=lib.eal:6 --log-level=lib.cryptodev:5 --log-level=user1:6 --iova-mode=pa --base-virtaddr=0x200000000000 --match-allocations --file-prefix=spdk0 --proc-type=auto ]
EAL: No available hugepages reported in hugepages-1048576kB
EAL: No legacy callbacks, legacy socket not created
; DBG:xnvme_be_spdk.c:_spdk_env_init-253: INFO: spdk_env_is_initialized: 1
; DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-743: INFO: trtype: PCIe, # 1 of 4
[2020-12-29 17:32:49.855961] nvme_pcie.c: 481:nvme_pcie_ctrlr_map_cmb: *ERROR*: get registers failed
[2020-12-29 17:32:49.856002] nvme_pcie.c: 855:nvme_pcie_ctrlr_construct: *ERROR*: get_cap() failed
[2020-12-29 17:32:49.856048] nvme.c: 486:nvme_ctrlr_probe: *ERROR*: Failed to construct NVMe controller for SSD: 0000:01:00.0
EAL: Requested device 0000:01:00.0 cannot be used
; DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-743: INFO: trtype: TCP, # 2 of 4
; DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-753: SKIP: !sys_uri
; DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-743: INFO: trtype: RDMA, # 3 of 4
; DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-753: SKIP: !sys_uri
# DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-743: INFO: trtype: FC, # 4 of 4
; DBG:xnvme_be_spdk.c:xnvme_be_spdk_enumerate-776: SKIP: enum. of trtype: FC not implemented
; DBG:xnvme_be_nosys.c:xnvme_be_nosys_enumerate-141: FAILED: not implemented(possibly intentional)
; DBG:xnvme_be.c:xnvme_enumerate-716: FAILED: fbsd->enumerate(...), err: 'Function not implemented', i: 2
xnvme_enumeration:
  capacity: 100
  nentries: 0
  entries: ~
[root@localhost ~]#

safl, 2021-01-04:
We are trying out Discord, feel free to join here:

https://discord.com/invite/XCbBX9DmKf

There we can chat in a more "synchronized" manner, which might be faster / more convenient for troubleshooting issues like this one. We can then update the issue here on GitHUB when we arrive at a solution.

surajk2, 2021-01-15:
Tried again with switch to uio_pci_generic. It worked. Thanks

I had upgraded my kernel to 5.3(from 3.1) too. So i would suggest to apply the solution to switching to uio_pci_generic first. If it does not work then try upgrading kernel.

closed 2021-01-15 by surajk2