xnvme/xnvme #8: Issue of linking libxnvme.a(built with spdk) with myproject

surajk2, 2021-01-15:
Context: I will try to explain the existing project setup and error encountered while linking libxnvme.a.
 

B (.so) 
libxnvme (.a)

A (other module .so)
X (other module .so)

A (.so) is linking B(.so)
X (.so) is linking B(.so)

1. I am working on B. B calls libxnvme API. Used linker flags as below for linking libxnvme(.a) in B. This linkage is required to use SPDK API.
;# LINKAGE
target_link_libraries(${B_NAME} -Wl,--whole-archive)
target_link_libraries(${B_NAME} -Wl,--no-as-needed)
target_link_libraries(${B_NAME} ${libxnvme.a})
target_link_libraries(${B_NAME} -Wl,--no-whole-archive)
target_link_libraries(${B_NAME} -Wl,--as-needed)

Result-> Twice linking error 
Duplicate method linker error observed with few spdk methods.

2. Workaround of issue mentioed in 1.
Used linker flags(same set of changes of 1) in the cmakelist of module A and removed from B. Able to use SPDK API.
This resolved twice linking error.

3. Now X complains that it is not finding xnvme_enumerate() in B.so
Had I been able to use #1 in first place (without twice linking error), I would not have used #2.
Error: ImportError: X.so: undefined symbol: xnvme_enumerate

4. To resolve error in #3, i tried putting same linker options in cmakelist of X as described in #1.
 It worked.
But this approach is making libxnvme APIs avialable/exposed in the modules where they are not required.

This approach is having problem as it will lead to exposing libxnvme API (alongwith spdk API) everywhere. 

A bit similar issue on stackoverflow: https://stackoverflow.com/questions/805555/ld-linker-question-the-whole-archive-option

Please feel free to ask more information in order to resolve the issue.

safl, 2021-01-15:
Hi @surajk2 ,
This is not that trivial to solve as SPDK  uses the constructor feature to register transports pcie/tcp/rdma. However, I will investigate a short-term work-around and a more viable long-term soluton.
Best,
Simon

surajk2, 2021-04-06:
@safl Can you please check and update about the status.

safl, 2021-04-06:
Hi @surajk2 , have not found a quick-fix for this issue. I expect to make new release of xNVMe by 16.04. Will provide another update by then.

surajk2, 2021-04-19:
Is there update ?

safl, 2021-04-19:
Hi @surajk2,
Release is pushed two weeks.

surajk2, 2021-04-19:
ok

safl, 2021-05-05:
Hi @surajk2, try ``v0.0.23`` out. A SPDK-workaround has been added so make sure you do:

```
git pull --rebase
git checkout v0.0.23
make clobber
git submodule update --init --recursive
make
sudo make install-deb
```

Then try your linking-configuration again. With workaround-patch then the SPDK NVMe PCIe Driver is loaded "manually" by xNVMe upon initialization of the SPDK/DPDK environment instead of relying on object ``.ctor`` / ``__attribute__((contructor))``/``--whole-archive`` registraiton of the driver/transport.

surajk2, 2021-05-06:
ok, i will try and update. Thanks

surajk2, 2021-05-19:
It works now. SPDK linker flag is not required while linking. Thanks:)

closed 2021-05-19 by surajk2