xnvme/xnvme #11: compiling faied

qihui81, 2021-07-06:
xNVMe-0.0.24/tests/ident.c:170:9: error: missing initializer for field ‘schm’ of ‘struct xnvme_ident’ [-Werror=missing-field-initializers]
  struct xnvme_ident ident = { 0 };

Looks like gcc doesn't support the initialization like struct xnvme_ident ident = { 0 };
use memset() can avoid the error

gcc version 8.3.1 20190311 (Red Hat 8.3.1-3) (GCC)

micvbang, 2021-07-06:
Hey,

It looks to me like you've set `-Werror=missing-field-initializers` yourself - is that correct?

xNVMe makes extensive use of this type of struct initialization (`struct type ident = { 0 }`), so I don't think this compiler flag makes sense :)

micvbang, 2021-07-06:
FYI: I've just confirmed that xNVMe master does build on a Debian Buster with `gcc (GCC) 8.3.0`

safl, 2022-02-18:
Hi @qihui81,

Assuming you got this fixed, for reference then the toolchains have been expanded to include:
openSUSE, CentOS Stream, Fedora, Alpine Linux, FreeBSD, and Windows:
https://github.com/OpenMPDK/xNVMe/actions/runs/1861022569

With text describing how to install here:
https://xnvme.io/docs/latest/getting_started/index.html#toolchain

Best,
Simon

closed 2022-02-18 by safl