xnvme/xnvme #17: Zoned-Namespaces (ZNS) Support in software/firmware level via backports

Daasin, 2021-12-08:
Is there a particular reason why something like this hasn't been ported in an update for users of older software and/or hardware? Since it was meant to be possible in theory to implement without dedicated hardware support of NVME 2.0 🤔  

and could an easier way  be brought forward in the future to use ZNS across multiple platforms if support is not directly allowed by specific sellers of the hardware and the SSD's that come with them? 🙏🏽

Daasin, 2022-03-02:
No worries :D 
and in both points i'm generally referring to the use of the NVME Zoned Namespaces Command Set (that exposes a zoned block storage interface for use by the Host System) 

From my understanding, it may not necessarily need a specific version of the NVME Specification nor hardware support and that one possible way of supporting it on SSD's is to simply do a firmware update. 
_**I essentially wanted to know, if it was possible to support across the board ...even if the manufacturer wasn't pushing out a firmware update.**_ Like a driver, or kernel module, or a new file system, or new volume management, changed partitioning scheme, or any other possible way that could be found rather than depending on individual manufacturers of SSD's like Samsung to support the feature in firmware updates.

safl, 2022-03-03:
To clarify, the Zoned Namespace Command Set is an NVMe specification; see [1]. Thus NVMe/ZNS is the interface exposing the SSD/device and its storage media to the host via, e.g., PCIe. NVMe provides other I/O command-set specifications:

* NVM Command Set[2], NVMe/NVM defines commands such as reading/writing logical blocks. If you have an M.2 NVMe SSD in your laptop/workstation, it most likely exposes the itself as an NVMe/NVM SSD. 
* Key Value Command Set[3], NVMe/KV defines commands for retrieving/storing key-value pairs. The NVMe/KV is a KV-database/object-storage packed inside an SSD. Pretty cool if you ask me :)

So, regarding **the thing you essentially want to know**:  the answer is no. Since, to turn an NVMe/NVM SSD into and NVMe/ZNS SSD, well, then you have to change it :)

You can of course use abstractions/shim-layers/emulations/in-directions in various layers of the software stack such that an abstraction presents itself as an NVMe/ZNS device although it is backed by NVMe/NVM, main-memory, tape or whatever, but that is just software-engineering.

One of these really useful abstractions are qemu, with qemu you can present an emulated/simulated NVMe/ZNS device to the guest, where the emulated device is a pure software abstraction backed by a file in a file-system, which could be on a NVMe/NVM or any other block device.

However, using NVMe/ZNS devices "natively" is where you reap the benefits, this is the approach of projects like:

- https://github.com/OpenMPDK/xZTL
- https://github.com/OpenMPDK/FlexAlloc
- https://github.com/westerndigitalcorporation/zenfs

The first two utilize xNVMe to unify software I/O path from application down to the device, for the first two one can switch between NVMe ioctl, libaio, io_uring, an user-space NVMe-drivers (SPDK NVMe) where-as the latter is tied to block-ioctl and pwrite/pread calls.

I hope the above answers your questions, otherwise, just ask some more.

Best,
Simon

References
=========

[1] https://nvmexpress.org/wp-content/uploads/NVM-Zoned-Namespace-Command-Set-Specification-1.1b-2022.01.05-Ratified.pdf
[2] https://nvmexpress.org/wp-content/uploads/NVM-Command-Set-Specification-1.0b-2021.12.18-Ratified.pdf
[3] https://nvmexpress.org/wp-content/uploads/NVM-Express-Key-Value-Command-Set-Specification-1.0b-2021.12.18-Ratified.pdf

safl, 2022-09-29:
Hi @Daasin. closing this now, feel free to open another issue with any other questions.

closed 2022-09-29 by safl