xnvme/xnvme #86: Unable to open device with fab URI

vkalef, 2022-05-16:
I'm working on version 0.0.29, over CentOS 7
1. Trying to open a device using the format fab:174.60.77.139:4420?nsid=1, but all the examples fails to get it as a valid format.
2. The --debug option on compile doesn't output the debug logs.

Help will be appreciated.

safl, 2022-07-08:
Hi @vkalef,
A lot has happened since v0.0.29, currently the latest integration of this is on this PR: https://github.com/OpenMPDK/xNVMe/pull/126
Can you try it, to see whether it fixes the issue?
Thanks!
Simon

karlowich, 2022-10-20:
Hi @vkalef, I'm closing this due to inactivity.
If you're still having trouble feel free to reopen, and I will be happy to help :)

closed 2022-10-20 by karlowich

vkalef, 2022-10-20:
We've managed to solve it, communication was done via discord, it just took
me a while to find out there is one.
I think you should emphasize it more on your site. It's very helpful.

karlowich, 2022-10-20:
That's great feedback!
Thanks @vkalef