xnvme/xnvme #109: Opening new namespace when device already open fails with SPDK backend

LiadOz, 2022-06-09:
We have started a discussion on this issue on Discord.
The issue happens on version 0.3.0 and I've confirmed it happens on centos 7 and ubuntu 20.

Reproduction steps:
setup nvmf app, create 2 devices and add only one to the subsystem
```
./build/bin/nvmf_tgt &
./scripts/rpc.py nvmf_create_transport -t TCP -u 16384 -m 8 -c 8192
./scripts/rpc.py nvmf_create_subsystem nqn.2016-06.io.spdk:cnode1 -a -s SPDK00000000000001 -d SPDK_Controller1
./scripts/rpc.py nvmf_subsystem_add_listener nqn.2016-06.io.spdk:cnode1 -t tcp -a  127.0.0.1 -s 4420
./scripts/rpc.py bdev_null_create Null0 8589934592 4096
./scripts/rpc.py bdev_null_create Null1 8589934592 4096
./scripts/rpc.py nvmf_subsystem_add_ns nqn.2016-06.io.spdk:cnode1 Null0
```

Run the following code (note the sleep)
```
#include <libxnvme.h>
#include <libxnvme_pp.h>
#include <unistd.h>

int
main(int argc, char *argv[])
{
        struct xnvme_opts opts1 = xnvme_opts_default();
        struct xnvme_opts opts2 = xnvme_opts_default();
        struct xnvme_dev *dev1;
        struct xnvme_dev *dev2;

        opts1.nsid = 1;
        opts2.nsid = 2;
        dev1 = xnvme_dev_open("127.0.0.1:4420", &opts1);
        sleep(20);
        dev2 = xnvme_dev_open("127.0.0.1:4420", &opts2);
        xnvme_dev_close(dev1);
        xnvme_dev_close(dev2);

        return 0;
}
```

When it starts sleeping add the other namespace
```
sudo ./scripts/rpc.py nvmf_subsystem_add_ns nqn.2016-06.io.spdk:cnode1 Null1
```

Here are the logs:
```
# DBG:xnvme_be.c:xnvme_be_factory-667: INFO: obtained backend instance be: 'spdk'
# DBG:xnvme_be_spdk_dev.c:_spdk_env_init-230: INFO: SPDK NVMe PCIe Driver registration -- BEGIN
# DBG:xnvme_be_spdk_dev.c:_spdk_env_init-235: INFO: skipping, already registered.
# DBG:xnvme_be_spdk_dev.c:_spdk_env_init-237: INFO: SPDK NVMe PCIe Driver registration -- END
[2022-06-09 13:20:24.558367] Starting SPDK v21.10 git sha1 4e4f11ff7 / DPDK 21.08.0 initialization...
[2022-06-09 13:20:24.558980] [ DPDK EAL parameters: [2022-06-09 13:20:24.559200] spdk [2022-06-09 13:20:24.559442] --no-shconf [2022-06-09 13:20:24.559667] -c 0x1 [2022-06-09 13:20:24.559891] --log-level=lib.eal:6 [2022-06-09 13:20:24.560124] --log-level=lib.cryptodev:5 [2022-06-09 13:20:24.560329] --log-level=user1:6 [2022-06-09 13:20:24.560526] --iova-mode=pa [2022-06-09 13:20:24.560754] --base-virtaddr=0x200000000000 [2022-06-09 13:20:24.560959] --match-allocations [2022-06-09 13:20:24.561153] --file-prefix=spdk_pid105448 [2022-06-09 13:20:24.561391] ]
EAL: No available 1048576 kB hugepages reported
TELEMETRY: No legacy callbacks, legacy socket not created
# DBG:xnvme_be_spdk_dev.c:_spdk_env_init-260: INFO: spdk_env_is_initialized: 1
# DBG:xnvme_be_spdk_dev.c:xnvme_be_spdk_state_init-798: ############################
# DBG:xnvme_be_spdk_dev.c:xnvme_be_spdk_state_init-799: INFO: trtype: PCIe, # 1 of 4
EAL: Cannot find device (0127:00:00.1)
EAL: Failed to attach device on primary process
[2022-06-09 13:20:24.702748] nvme.c: 838:nvme_probe_internal: *ERROR*: NVMe ctrlr scan failed
[2022-06-09 13:20:24.702767] nvme.c: 915:spdk_nvme_probe: *ERROR*: Create probe context failed
# DBG:xnvme_be_spdk_dev.c:xnvme_be_spdk_state_init-810: FAILED: probe a:0, e:-1, i:0
# DBG:xnvme_be_spdk_dev.c:xnvme_be_spdk_state_init-798: ############################
# DBG:xnvme_be_spdk_dev.c:xnvme_be_spdk_state_init-799: INFO: trtype: TCP, # 2 of 4
trid:
  trstring: 'TCP'
  trtype: 0x3
  adrfam: 0x1
[... 178 lines cut ...]
				return;
			}
			nvme_ctrlr_parse_ana_log_page(ctrlr, nvme_ctrlr_update_ns_ana_states,
						      ctrlr);
		}
	}

	active_proc = nvme_ctrlr_get_current_process(ctrlr);
	if (active_proc && active_proc->aer_cb_fn) {
		active_proc->aer_cb_fn(active_proc->aer_cb_arg, cpl);
	}
}
```
it is defined in nvme_ctrlr.c

To solve it on our end we created a workaround (that is probably not a good one) in xnvme_be_spdk_state_init, when probing fails on max attempt we call these functions
``` 
nvme_ctrlr_clear_changed_ns_log(ctrlr);
nvme_ctrlr_identify_active_ns(ctrlr);
nvme_ctrlr_update_namespaces(ctrlr);
nvme_io_msg_ctrlr_update(ctrlr);
```
then we run the function again.

@safl what do you think should be our approach to solve this?

safl, 2022-06-09:
Hi @LiadOz,
Thanks for providing all the information. This is somewhat related to the other issue of xNVMe reference-counting and re-using the controller instance. A possible immediate approach would be to use ``xnvme_enumerate()`` to obtain handles, instead of ``xnvme_dev_open()``. See: https://xnvme.io/docs/latest/capis/xnvme.html#xnvme-enumerate

This is not well-documented, but for ``xnvme_enumerate()`` for the callback-function passed to it, if it returns ``XNVME_ENUMERATE_DEV_KEEP_OPEN``, then the enumerated handle can be used instead of calling ``xnvme_dev_open()`` multiple times. The callback-function can then put the handle on a list via the ``**cb_args``.

However, in the case where the namespaces are added later, then this of course would not suffice. One solution comes to mind, on the xNVMe side. Change: ``xnvme_be_spdk_dev.c``to not reference-count and not re-use attached controllers. This is needed for PCIe, however, for fabrics it should not be required. By doing so, the stale state should not be a problem.
I did consider calling the functions your mentioned, but afaik, these are not part of the public SPDK API, but only exposed internally in the SPDK ``nvme_ctrl.c`` module.

**Note:** Another thing xNVMe most likely should provide, is a means to register a callback for handling these events. Either, managing it in the backend or providing user-control over it.

safl, 2022-07-08:
There were several things preventing this to work properly, I have tried integrating the changes here:
https://github.com/OpenMPDK/xNVMe/pull/126
Can you try it out?
Thanks!

karlowich, 2022-10-20:
Hi @LiadOz, I have tried to reproduce this but I don't see the error you're describing.
Can you check if this is still a problem on `next`?
Thanks!

LiadOz, 2022-10-20:
This was solved in #114, Thanks

closed 2022-10-20 by LiadOz