xnvme/xnvme #127: SPDK backend fails enumeration

Baekalfen, 2022-07-13:
SPDK has an issue during enumeration, when it calls `xnvme_dev_open` here: https://github.com/OpenMPDK/xNVMe/blob/e507f2d1791163d0c2757f1b973cd2cf6dfcb573/lib/xnvme_be_spdk_dev.c#L611

It doesn't force `spdk` backend in the `opts` variable, therefore you may trigger `vfn` or another PCIe backend to try to open the device. This can happen if `spdk` fails to open the device, and `xnvme_dev_open` continues to probe other bakcends.

safl, 2022-07-13:
This PR: https://github.com/OpenMPDK/xNVMe/pull/126
Should fix this for SPDK.

However, the other backends exhibit the same behavior, and probably need similar adjustments as done for SPDK on PR #126.

safl, 2022-10-06:
Closed without merge: https://github.com/OpenMPDK/xNVMe/pull/126
This needs some renewed attention.

karlowich, 2022-11-10:
This has been resolved by #190

PR #190 (2022-11-07): be: add earmark to dev_open() during enumeration
Pass a copy of opts, with the respective backend set, to xnvme_dev_open().
This is to avoid a case where a different backend is used for
enumeration and opening the device.

Closes #127
closed 2022-11-10 by karlowich