xnvme/xnvme #130: xnvme_queue_poke hangs when connection is dropped in spdk backend

LiadOz, 2022-07-17:
When I try to make io commands while the connection to the spdk device is dropped the function xnvme_queue_poke hangs, this happens in the spdk backend but not when using nvme-cli. The behavior I expect is that the function should return an error that I would be able to handle on my side.

when using nvme-cli this is what I get in the debug log
```
# DBG:xnvme_be_linux_nvme.c:ioctl_wrap-137: INFO: ioctl(NVME_IOCTL_IO64_CMD), err(-1), errno(11)
# DBG:xnvme_be_linux_nvme.c:ioctl_wrap-144: INFO: retconv: -1 and set errno
# DBG:xnvme_be_linux_nvme.c:xnvme_be_linux_nvme_cmd_io-255: FAILED: ioctl_wrap(), err: -11
# DBG:xnvme_be_posix_async_emu.c:_posix_async_emu_poke-136: FAILED: sync.cmd_io{v}(), err: -11
# ERR: 'got completion errors': {errno: 5, msg: 'Input/output error'}
```

Here is how to reproduce the issue:
1) create spdk device
```
./build/bin/nvmf_tgt &
./scripts/rpc.py nvmf_create_transport -t TCP -u 16384 -m 8 -c 8192
./scripts/rpc.py nvmf_create_subsystem nqn.2016-06.io.spdk:cnode1 -a -s SPDK00000000000001 -d SPDK_Controller1
./scripts/rpc.py nvmf_subsystem_add_listener nqn.2016-06.io.spdk:cnode1 -t tcp -a  127.0.0.1 -s 4420
./scripts/rpc.py bdev_null_create Null0 8589934592 4096
./scripts/rpc.py nvmf_subsystem_add_ns nqn.2016-06.io.spdk:cnode1 Null0
```

2) run examples/xnvme_single_async (change opts.nsid to 1) and stop with a breakpoint at line 96 (before issuing the command)
3) configure iptables to drop incoming connection
```
sudo iptables -I INPUT -p tcp -i lo --dport 4420 -j DROP
```
Then xnvme_queue_drain should hang

To restore the ip tables run:
```
sudo iptables -D INPUT -p tcp -i lo --dport 4420 -j DROP
```

LiadOz, 2022-08-01:
Actually xnvme_queue_poke does not hang, xnvme_queue_drain hangs because there is no result to the tasks in the spdk transport queue.
I have pushed a fix we use to handle this issue here #135 In this fix I registered a callback function for timeouts using **spdk_nvme_ctrlr_register_timeout_callback**, I did not see a suitable function to call to abort certain commands so I created one in spdk. Also it seems that the function that checks if for a timeout in spdk is not always called. We sometimes got SIGPIPE signals when spdk_sock_flush so we added **MSG_NOSIGNAL** to the send function (this is fixed in a later version of spdk).
I will try to see if we could integrate these changes directly into spdk.

LiadOz, 2022-08-16:
It was suggested to me on spdk slack to reconnect on timeout instead of using the function I've created here #135 `nvme_transport_qpair_finish_req`, unfortunately there is no async reconnect on the version of spdk used right now on xNVMe, so for the time being I am failing the controller on any command timeout.
I have also submitted a MR to spdk that fixes issues with timeout https://github.com/spdk/spdk/issues/2614, this is patch I added on #135

safl, 2022-09-29:
Hi @LiadOz , SPDK is currently at v22.05, when v22.09 is out (shortly) I will update the subproject and the out-of-tree patch should no longer be needed.

karlowich, 2022-11-28:
Hi @LiadOz, 
After digging around in the spdk documentation for a while, I found that the simplest solution to this problem was to set `transport_ack_timeout` in the `spdk_nvme_ctrlr_opts`.
I am unsure which interval is the optimal for the timeout. I set it to 2^16 ms, but that might be too long.
What do you think? 

Please checkout #216 and let me know if this solves the problem for you.

LiadOz, 2022-11-28:
It seems that this solves the basic examples I've had, we will check it on our environment with more complicated tests to make sure it resolves the issue.
Regarding the timing, [here](https://elixir.bootlin.com/linux/v5.19.12/source/drivers/nvme/host/core.c#L38:~:text=unsigned%20int%20admin_timeout%20%3D%2060%3B%0Amodule_param(admin_timeout%2C%20uint%2C%200644)%3B%0AMODULE_PARM_DESC(admin_timeout%2C%20%22timeout%20in%20seconds%20for%20admin%20commands%22)%3B%0AEXPORT_SYMBOL_GPL(admin_timeout)%3B%0A%0Aunsigned%20int%20nvme_io_timeout%20%3D%2030%3B) you can see the timeout defined for nvme tcp in the linux kernel, 30 seconds for regular commands and 60 seconds for admin commands, I guess that if a global timeout is going to be used, then it better be 60 seconds.

karlowich, 2022-12-05:
What is the status on this? Does #216 resolve your issue?

LiadOz, 2022-12-05:
I had an issue in some of the runs, I only stopped poking on -6 return status and it seems that either -6 or -1 could be returned from poke. I will try running it again.

LiadOz, 2022-12-08:
@karlowich Your solution does work for the example I have given and solves most of the timeout issues I have as using `spdk_nvme_ctrlr_register_timeout_callback`. But there are still cases where the issue persists, and unfortunately I did not find a way to simply reproduce this issue. This is because the two solutions are fundamentally different. By using `spdk_nvme_ctrlr_register_timeout_callback` we can guarantee that any command sent will finish in a given time. The second solution guarantees that if there is no ack response for the device then the  poke functions will return an error.

Here is an example scenario where linux kernel and xnvme spdk backend behaves differently:
A device is malfunctioning and stops processing commands, meaning it will never notify the user that a command has finished. If I were to call `xnvme_queue_drain` with kernel backend, the command will timeout after 30 seconds. Had I used spdk backend `xnvme_queue_drain` would hang forever.

Since I am still having an issue, and the device I am using this on seems to not have the issue when used with nvme-cli, I believe that there are more sensible cases in which you would still need to use a solution like `spdk_nvme_ctrlr_register_timeout_callback`

karlowich, 2022-12-12:
I am closing this, since #135 is now merged.

closed 2022-12-12 by karlowich