xnvme/xnvme #134: When is libkvs available

diluga, 2022-08-01:
Is there any news about libkvs?

safl, 2022-08-17:
Hi @diluga,
We currently have an implementation, release of it is post-poned as we currently have no "upstream" way of testing it.
To remedy this, we are working on providing support in qemu and using that as the test-vehicle similar to what we do for the other NVMe-specific features requiring device-support.
Best,
Simon

Ignalina, 2022-09-24:
@safl
Does NVMe 2.0 "contain" kv api ?  Just ordered an Samsung 990 Pro on amazon that is specified by the "seller" to be nvme 2.0 but I havent figured out if it contains kv api .. but would love to test the libkvs if/when possible.

/Rickard

Just ordered one from Amazon
 Ive just ordered the Samsung 990 PRO from amazon ,

safl, 2022-10-04:
Hi @Ignalina / Rickard,

Command-set support is optional, with NVMe 2.0 multiple command-sets are defined, you can see a list [here](https://nvmexpress.org/developers/nvme-command-set-specifications/), these include:

* The NVM Command-Set (these are your traditional read/write commands)
* The Zoned Namespace Command-Set (NVM + management send/recv + optional append)
* The KV Command-Set (list, exists, store/retrieve commands etc.)

Most client SSDs, such as the one you referred to only support the NVM Command-Set.

Ignalina, 2022-10-04:
Many thanks @safl / Simon I will enjoy my Samsung 990 Pro in other ways then :) 

I corrected my "software-hardware survey"  in https://www.x14.se/to-a-brighter-future/   , thanks again.

safl, 2022-10-19:
> 

You are most welcome :)

karlowich, 2023-05-30:
Hi @diluga, now the patch for KV support (#270) has been merged and is available on the `next` branch.

safl, 2023-06-26:
Hi @diluga ,KV Support is now part of the latest release (v0.7.0), thus closing this issue.

closed 2023-06-26 by safl

Neurone, 2023-11-08:
Hi @safl @Baekalfen @karlowich thanks for the clarification and the development of this feature, it's very cool and I'm very keen to test it on real HW.

How can I check/verify an SSD drive implement a specific Command Set? How do you discovered the 990 Pro does not implement it? Do you know some SSD drive that does it?

@Ignalina I fell in the same misunderstanding about the Samsung 990 Pro: when I read "NVMe 2.0" I considered KV Command-Set enabled by default, so bought it just a month ago and I tried to make it work with xNVMe... and now I understand why it was not working at all, regardless of my tries :_(