xnvme/xnvme #157: Deprecated documentation

safl, 2022-09-19:
The documentation needs a read through for deprecated things such as:

https://xnvme.io/docs/latest/getting_started/index.html#device-identifiers

Which has an example:
```
...
struct xnvme_dev *dev = xnvme_dev_open("pci:0000:01:00.0?nsid=1");
...
```

This is the old device-uri format which has been deprecated in favor of ``struct xnvme_opts``.
This should be fixed to avoid confusion.

niclashedam, 2022-11-30:
I can spend som time on it sometime soon if you like.

safl, 2023-02-24:
@niclashedam did not see that you responded here. That would be excellent, if you still have the time :)

niclashedam, 2023-02-24:
My backlog for December through February ended up being more significant than anticipated. I will look into it as soon as possible.

niclashedam, 2023-07-25:
I have had a quick read-through of parts of the documentation, and for many of the pages, I need more experience with xNVME to judge whether it's up to date. Can I use any strategy to determine if it's up to date?

karlowich, 2024-04-08:
Hi @niclashedam, sorry we missed your reply!

I think you are indeed right, that finding outdated parts of the documentation requires intimate knowledge of xNVMe.
If you still want to contribute in this area, you can help out by reading through the "C API" section (and the examples) in the latest version of the [documentation](https://xnvme.io/docs/next/capis/index.html).

My goal is to make it easier to get started with xNVMe and understand which parts of the API you absolutely need, and which are "optional".

karlowich, 2024-06-14:
Closing this, as it is now tracked here: #475

closed 2024-06-14 by karlowich