xnvme/xnvme #206: Deprecation-warnings on MacOS

safl, 2022-11-15:
In addition to format-specifiers giving build-warnings (this is addressed here: https://github.com/OpenMPDK/xNVMe/pull/205), then the following deprecation-warnings and additional warnings need some attention:

```
[24/202] Compiling C object lib/libxnvme-shared.dylib.p/xnvme_be_macos_dev.c.o
../lib/xnvme_be_macos_dev.c:33:42: warning: 'kIOMasterPortDefault' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
        matching_dictionary = IOBSDNameMatching(kIOMasterPortDefault, 0, basename(devname));
                                                ^~~~~~~~~~~~~~~~~~~~
                                                kIOMainPortDefault
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:133:19: note: 'kIOMasterPortDefault' has been explicitly marked deprecated here
const mach_port_t kIOMasterPortDefault
                  ^
../lib/xnvme_be_macos_dev.c:34:49: warning: 'kIOMasterPortDefault' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
        ioservice_device = IOServiceGetMatchingService(kIOMasterPortDefault, matching_dictionary);
                                                       ^~~~~~~~~~~~~~~~~~~~
                                                       kIOMainPortDefault
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:133:19: note: 'kIOMasterPortDefault' has been explicitly marked deprecated here
const mach_port_t kIOMasterPortDefault
                  ^
../lib/xnvme_be_macos_dev.c:78:7: warning: incompatible pointer types passing 'IONVMeSMARTInterface ***' (aka 'struct IONVMeSMARTInterface ***') to parameter of type 'LPVOID *' (aka 'void **') [-Wincompatible-pointer-types]
                                         nvme_smart_interface);
                                         ^~~~~~~~~~~~~~~~~~~~
../lib/xnvme_be_macos_dev.c:178:19: warning: incompatible pointer to integer conversion passing 'IONVMeSMARTInterface **' (aka 'struct IONVMeSMARTInterface **') to parameter of type 'io_object_t' (aka 'unsigned int') [-Wint-conversion]
                IOObjectRelease(state->nvme_smart_interface);
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:268:14: note: passing argument to parameter 'object' here
        io_object_t     object );
                        ^
../lib/xnvme_be_macos_dev.c:262:27: warning: incompatible pointer to integer conversion assigning to 'io_object_t' (aka 'unsigned int') from 'void *' [-Wint-conversion]
                state->ioservice_device = NULL;
                                        ^ ~~~~
5 warnings generated.
[94/202] Compiling C object lib/libxnvme-static.a.p/xnvme_be_macos_dev.c.o
../lib/xnvme_be_macos_dev.c:33:42: warning: 'kIOMasterPortDefault' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
        matching_dictionary = IOBSDNameMatching(kIOMasterPortDefault, 0, basename(devname));
                                                ^~~~~~~~~~~~~~~~~~~~
                                                kIOMainPortDefault
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:133:19: note: 'kIOMasterPortDefault' has been explicitly marked deprecated here
const mach_port_t kIOMasterPortDefault
                  ^
../lib/xnvme_be_macos_dev.c:34:49: warning: 'kIOMasterPortDefault' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
        ioservice_device = IOServiceGetMatchingService(kIOMasterPortDefault, matching_dictionary);
                                                       ^~~~~~~~~~~~~~~~~~~~
                                                       kIOMainPortDefault
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:133:19: note: 'kIOMasterPortDefault' has been explicitly marked deprecated here
const mach_port_t kIOMasterPortDefault
                  ^
../lib/xnvme_be_macos_dev.c:78:7: warning: incompatible pointer types passing 'IONVMeSMARTInterface ***' (aka 'struct IONVMeSMARTInterface ***') to parameter of type 'LPVOID *' (aka 'void **') [-Wincompatible-pointer-types]
                                         nvme_smart_interface);
                                         ^~~~~~~~~~~~~~~~~~~~
../lib/xnvme_be_macos_dev.c:178:19: warning: incompatible pointer to integer conversion passing 'IONVMeSMARTInterface **' (aka 'struct IONVMeSMARTInterface **') to parameter of type 'io_object_t' (aka 'unsigned int') [-Wint-conversion]
                IOObjectRelease(state->nvme_smart_interface);
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:268:14: note: passing argument to parameter 'object' here
        io_object_t     object );
                        ^
../lib/xnvme_be_macos_dev.c:262:27: warning: incompatible pointer to integer conversion assigning to 'io_object_t' (aka 'unsigned int') from 'void *' [-Wint-conversion]
                state->ioservice_device = NULL;
```

niclashedam, 2022-11-15:
On it 👍

safl, 2022-11-15:
> On it 👍

Awesome @niclashedam !

safl, 2022-11-25:
Thanks @niclashedam

PR #212 (2022-11-25): Fix macOS warnings
This PR resolves #206 and the code is based on documentation of macOS Ventura 13.0 (22A380) and compiles successfully on said device.
closed 2022-11-25 by safl