xnvme/xnvme #238: Build system does not properly build out-of-tree for subprojects

metaspace, 2023-01-25:
When building xnvme with

```shell
meson setup </path/to/build> </path/to/xnvme>
meson compile -C </path/to/build>
```

changes are made to downloaded artifacts in the subprojects directory of the xnvme codebase. If the codebase is moved or mounted in a container, invocation of `meson setup` will fail if the xnvme code base lives on a different path in the container:

```text
Message: Checking if generic platform (Apple Silicon)
Message: Building ..
Program make found: YES (/usr/bin/make)
Running command: /usr/bin/make -j 8
--- stdout ---

--- stderr ---
/mnt/xnvme/subprojects/spdk/mk/spdk.common.mk:269: /home/aeh/src/xnvme/xnvme/subprojects/spdk/lib/env_dpdk/env.mk: No such file or directory
make: *** No rule to make target '/home/aeh/src/xnvme/xnvme/subprojects/spdk/lib/env_dpdk/env.mk'.  Stop.

../mnt/xnvme/subprojects/spdk/meson.build:33:2: ERROR: Command "/usr/bin/make -j 8" failed with status 2.
```

The path `/home/aeh/src/xnvme/xnvme/subprojects/spdk/lib/env_dpdk/env.mk` does not exist inside the container. It is leftover state from when xnvme was compiled outside the container.

Subproject components should be placed under `</path/to/build>` to avoid this issue.

safl, 2023-02-24:
Sorry for the slow response on this. As you have observed then the use of the SPDK subprojects does not operate "cleanly". The main challenge here is that we really want it to be "easy" to build and install xNVMe with all dependencies. However, for SPDK, which is not meson, doing so will yield issues such as the scenario you present here, and special handling is thus required.

The path xNVMe is taking here is going towards more conventional approaches. In previous version of xNVMe liburing was bundlded in a manner similar to SPDK. Liburing was the first thing to be removed from subprojects and library-bundling. Next one up is "fio", since we now have the xNVMe I/O engine in upstream fio, we now await the next release of fio, then it can be removed from the subproject and custom-build.
After that, then SPDK will be removed, which is WIP: https://github.com/OpenMPDK/xNVMe/pull/243

Last remaining is libvfn, which might get to stick around, since it is meson, it behaves as it should and can easily be disabled.

safl, 2023-02-24:
This issues catches the items: https://github.com/OpenMPDK/xNVMe/issues/164