xnvme/xnvme #246: win10 install fail

Jojo-moon, 2023-02-16:
[image]
[image]

rs-sam, 2023-02-16:
Hi @Jojo-moon, Thanks for posting your query here.
I can see meson is using 'gcc' compiler but xNVMe-windows is based on clang compiler. Please refer to the 'toolbox/pkgs/windows-2019.bat' script to ensure that required build toolchain is installed on your machine.

karlowich, 2023-02-17:
Hi @Jojo-moon, 
It seems you have found a bug in our build system, thanks!
I have created a PR to fix the bug, please try it out: #247 

However, please also follow the advice of @rs-sam :)

Jojo-moon, 2023-02-17:
I had tried #247 ,However, errors will still be reported when installing fio. So i ask for help.

[image]

Jojo-moon, 2023-02-20:
> I had tried #247 ,However, errors will still be reported when installing fio. So i ask for help.
> 
> [image]

I had tried https://github.com/OpenMPDK/xNVMe/pull/247 ,However, errors will still be reported when installing fio. So i ask for help.

061ff7c1-4855-4459-8941-4438a52085cb

Jojo-moon, 2023-02-20:
> Hi @Jojo-moon, It seems you have found a bug in our build system, thanks! I have created a PR to fix the bug, please try it out: #247
> 
> However, please also follow the advice of @rs-sam :)

I had tried https://github.com/OpenMPDK/xNVMe/pull/247 ,However, errors will still be reported when installing fio. So i ask for help.

061ff7c1-4855-4459-8941-4438a52085cb

rs-sam, 2023-02-20:
Hi @Jojo-moon 
Have you tried running 'toolbox/pkgs/windows-2019.bat' batch on your machine? It will install the required tools and packages.

If you are facing any issue, packages can be installed manually using MSYS2 terminal.
1. C compiler for the host machine should be 'clang'
   clang package can be installed using 'pacman -S mingw-w64-x86_64-clang' under MSYS2 terminal.
2. FIO build system for windows is based on cygwin
    chocolatey based 'make.exe' is build for windows, to build FIO MSYS based 'make.exe' will be required to do cross- 
    compilation.
    make package can be installed using 'pacman -S mingw-w64-x86_64-make' under MSYS2 terminal.

Adding a screenshot for your reference, to see the tools and its paths, 
[image]

Jojo-moon, 2023-02-20:
> Hi @Jojo-moon Have you tried running 'toolbox/pkgs/windows-2019.bat' batch on your machine? It will install the required tools and packages.
>
> If you are facing any issue, packages can be installed [...]

I had run windows-2019.bat always. I had do following steps :

Set-ExecutionPolicy Bypass -Scope Process -Force; [System.Net.ServicePointManager]::SecurityProtocol = [System.Net.ServicePointManager]::SecurityProtocol -bor 3072; iex ((New-Object System.Net.WebClient).DownloadString('https://community.chocolatey.org/install.ps1'))

   cd toolbox\pkgs
   windows-2019.bat
   cd xnvme
   meson setup builddir (failed as above)

Jojo-moon, 2023-02-21:
> Hi @Jojo-moon Have you tried running 'toolbox/pkgs/windows-2019.bat' batch on your machine? It will install the required tools and packages.
> 
> If you are facing any issue, packages can be installed manually using MSYS2 terminal.
> 
> 1. C compiler for the host machine should be 'clang'
>    clang package can be installed using 'pacman -S mingw-w64-x86_64-clang' under MSYS2 terminal.
> 2. FIO build system for windows is based on cygwin
>    chocolatey based 'make.exe' is build for windows, to build FIO MSYS based 'make.exe' will be required to do cross-
>    compilation.
>    make package can be installed using 'pacman -S mingw-w64-x86_64-make' under MSYS2 terminal.
> 
> Adding a screenshot for your reference, to see the tools and its paths, [image]
> 
> Note: Use 'build.bat' in xnvme folder to start complilation.

I had built xnvme successfully by pulling next branch code ,thanks!

safl, 2023-02-27:
Thanks @karlowich and @rs-sam for helping out @Jojo-moon !
@Jojo-moon looks like your issue is resolved. In case it is not, then feel free to re-open.

closed 2023-02-27 by safl