xnvme/xnvme #248: fio use external io engine fail on ubuntu 20.04

Jojo-moon, 2023-02-17:
[image]

By the way,does xnvme support Android system？

safl, 2023-02-17:
This happens when using the external xnvme io-engine with a version of fio which has support for the internal xnvme io-engine. So you can choose:

* instead of using the external engine, then use the internal one
* build fio without the internal engine and then use the external engine.

safl, 2023-02-17:
We are not testing on Android, but i do not see a reason why xnvme should not work ln Android.

Jojo-moon, 2023-02-20:
> We are not testing on Android, but i do not see a reason why xnvme should not work ln Android.

> This happens when using the external xnvme io-engine with a version of fio which has support for the internal xnvme io-engine. So you can choose:
> 
> * instead of using the external engine, then use the internal one
> * build fio without the internal engine and then use the external engine.

I don't quite understand. I have the following confusion.

1. I installed xnvme, but the command line could not find fio. I installed fio additionally before, but I see that xnvme has subproject fio. Do you know if xnvme has installed fio? Do I need to install fio additionally?

2. fio has supported internal xnvme io engine. What does it mean?

ankit-sam, 2023-02-21:
Hi @Jojo-moon

We have both the external and internal xnvme fio engine
- The external fio engine is any engine which is not bundled with the fio source code. The external xnvme fio engine is present in xNVMe repo at tools/fio-engine/xnvme_fioe.c.
- Any internal fio engine is the one which is there with the fio source code. The internal xnvme fio engine is present in fio repo at engines/xnvme.c

what @safl mentioned is that:

[2]. you can try and use external fio engine, and it will require you to disable the internal fio engine.
Command to disable internal fio engine and building fio is:

```./configure --disable-xnvme```
```make && make install```

[1]. or you can use the internal fio engine, sample configuration file can be found here:
https://github.com/axboe/fio/blob/master/examples/xnvme-compare.fio

We will soon remove the external xnvme fio engine, and are currently waiting for a newer version of fio release.

karlowich, 2023-03-13:
Hi @Jojo-moon, is this solved?
Then I will close this issue.

closed 2023-06-07 by karlowich