xnvme/xnvme #252: CI fails on Debian Bookworm

safl, 2023-02-23:
Debian Bookworm seems to be "forcing" users to install packages in a virtual-environment.
Atleast, something changed in Bookworm yielding the error:

```
Run python3 -m pip install xnvme-core/dist/xnvme-core-*.tar.gz --user
WARNING: The directory '/github/home/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you should use sudo's -H flag.
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error: Process completed with exit code 1.
```

karlowich, 2023-02-24:
This seems to be the explanation:
> ## 5.2.2. Python Interpreters marked externally-managed
> The Debian provided python3 interpreter packages (python3.11 and pypy3) are now marked as being externally-managed, following [PEP-668](https://peps.python.org/pep-0668/). The version of python3-pip provided in Debian follows this, and will refuse to manually install packages on Debian's python interpreters, unless the --break-system-packages option is specified.

> If you need to install a Python application (or version) that isn't packaged in Debian, we recommend that you install it with pipx (in the pipx Debian package). pipx will set up an environment isolated from other applications and system Python modules, and install the application and its dependencies into that.

> If you need to install a Python library module (or version) that isn't packaged in Debian, we recommend installing it into a virtualenv, where possible. You can create virtualenvs with the venv Python stdlib module (in the python3-venv Debian package) or the virtualenv Python 3rd-party tool (in the virtualenv Debian package). For example, instead of running pip install --user foo, run: mkdir -p ~/.venvs && python3 -m venv ~/.venvs/foo && ~/.venvs/foo/bin/python -m pip install foo to install it in a dedicated virtualenv.

> See /usr/share/doc/python3.11/README.venv for more details.

From the [release notes](https://www.debian.org/releases/bookworm/amd64/release-notes/ch-information.en.html#python3-pep-668)

safl, 2023-03-15:
Fix on https://github.com/OpenMPDK/xNVMe/pull/257

safl, 2023-03-17:
Closing as it is fixes on #257 , keeping it in mind for #219

closed 2023-03-17 by safl