xnvme/xnvme #258: win10 build fail sometimes and install fail

Jojo-moon, 2023-03-08:
1 .run build.bat will  always hold  :
[image]

2. run build.bat install fail
[image]

karlowich, 2023-03-10:
Hi @Jojo-moon, regarding your points:
1. This is perfectly fine

2. This is a problem, yes. However, your screenshot makes it hard to see what went wrong. Which branch are you on?

3. It is expected that you have to add to your PATH before being able to run xnvme commands.
In the CI we add these to PATH:
```set "PATH=%SystemDrive%\tools\msys64;%PATH%"
  set "PATH=%SystemDrive%\tools\msys64\usr\bin;%PATH%"
  set "PATH=%SystemDrive%\tools\msys64\mingw64\bin;%PATH%"
  ```
  
4. This is likely a problem with your `cmd-out.nvmec` and not xNVMe. 
The Microsoft NVMe driver doesn't support pass-through.
@rs-sam do you have a solution for this?

Jojo-moon, 2023-03-13:
> Hi @Jojo-moon, regarding your points:
>
> 1. This is perfectly fine
> 2. This is a problem, yes. However, your screenshot makes it hard to see what went wrong. Which branch are you on?
> 3. It is expected [...]

1. on main branch ,however next branch may be the same.
2.  I just run test_padc with opcode=0xc1,others not change.  I tried test padc with opcode=0x6 ,it works!

rs-sam, 2023-03-13:
Hi @Jojo-moon 
Current xNVMe Windows backend does not support pass-through commands.
There is a new option called 'STORAGE_PROTOCOL_COMMAND' that can be used to send vendor specific admin/nvme commands, xNVMe build system is based on MinGW headers and libs and this 'STORAGE_PROTOCOL_COMMAND' is not yet available in that.

Jojo-moon, 2023-03-13:
> Hi @Jojo-moon Current xNVMe Windows backend does not support pass-through commands. There is a new option called 'STORAGE_PROTOCOL_COMMAND' that can be used to send vendor specific admin/nvme [...]

I I konw STORAGE_PROTOCOL_COMMAND can be used to send vs command . Will xnvme support it later ?

rs-sam, 2023-03-13:
Yes, it is in our bucket-list.

closed 2023-06-07 by karlowich