xnvme/xnvme #304: Misleading error message for nvme info.

sbates130272, 2023-07-21:
When one tries to get xnvme info on a device node that one does not have appropriate permissions for the error message returned seems misleading in the sense it states the file is not there. Would a more appropriate message not be to inform the user the file is there but they lack permissions to open it?

```
batesste@xnvme-test-qemu-1:~$ xnvme info /dev/nvme0
D nvme_init (src/nvme/core.c:514): nvme/core: could not get device class code
# ERR: 'xnvme_dev_open()': {errno: -2, msg: 'No such file or directory'}
batesste@xnvme-test-qemu-1:~$ sudo xnvme info /dev/nvme0
xnvme_dev:
  xnvme_ident:
    uri: '/dev/nvme0'
    dtype: 0x4
    nsid: 0x1
    csi: 0x1f
    subnqn: ''
  xnvme_be:
    admin: {id: 'file_as_ns'}
    sync: {id: 'psync'}
    async: {id: 'emu'}
    attr: {name: 'linux'}
  xnvme_opts:
    be: 'linux'
    mem: 'posix'
    dev: 'FIX-ID-VS-MIXIN-NAME'
    admin: 'file_as_ns'
    sync: 'psync'
    async: 'emu'
    oflags: 0x4
  xnvme_geo:
    type: XNVME_GEO_CONVENTIONAL
    npugrp: 1
    npunit: 1
    nzone: 1
    nsect: 1
    nbytes: 1
    nbytes_oob: 0
    tbytes: 0
    mdts_nbytes: 65024
    lba_nbytes: 512
    lba_extended: 0
    ssw: 9
```

karlowich, 2023-07-25:
Hi @sbates130272, I can reproduce your error and I agree that it is not ideal.
It comes from the way xNVMe handles backends.
When a backend is not specified, xNVMe will try every backend to see if any of them work.
In this case, none of the backends can open the device, but for different reasons.
SPDK and the ramdisk give errno 22 (invalid argument) since the device name is not of the format they expect.
Linux correctly gives errno 13 (permission denied).
VFIO gives errno 2 (file not found), which should maybe be errno 22 instead, but that is besides the point.
xNVMe has no way of knowing which error is the most applicable, so we just return the latest error we have seen.
Due to the somewhat arbitrary ordering of the backends, the latest error comes from VFIO and "file not found" is reported to the user.
To see all of this, you can compile xNVMe with debug enabled by using `make config-debug`.

safl, 2023-08-14:
This is an area that needs some attention. The error-handling should be normalized / unified somehow. Adding this to backlog.