xnvme/xnvme #307: suggestions for xnvme py library loader

jwdevantier, 2023-07-24:
Two suggestions for making the library loader (even) better

**1 inspirational example, (CL)-cffi**
```lisp
  (define-foreign-library libcurl
    (:darwin (:or "libcurl.3.dylib" "libcurl.dylib"))
    (:unix (:or "libcurl.so.3" "libcurl.so"))
    (t (:default "libcurl")))
   
  (use-foreign-library libcurl)
```

This snippet attempts to load the curl library. If the platform is macOS (darwin), first try to load "libcurl.3.dylib", if that fails, try loading "libcurl.dylib".
If it is unix (Linux, BSD, ...) try "libcurl.3.so" before "libcurl.so".
Finally, for every other platform, try "libcurl", but add the platform-specific extension (.dll for windows, .so for linux, ...) to the name.

**2 make library lookup easy to override**
Recognizing that even the best attempts may be in vain, the https://pypi.org/project/clang/ library allows you
to call `clang.cindex.Config.set_library_file` to explicitly load a specific library file, or `clang.cindex.Config.set_library_path()` to provide an additional directory in which to search for the clang library in.

This is a good idea, except it requires downstream projects to *also* expose that method somehow, which ctypeslib2 does not (for instance).

A better approach would be for the loader to error out as it does now ( https://github.com/OpenMPDK/xNVMe/blob/5198563801b14d21d801a7224af63b9e85d798a3/python/bindings/auxiliary/patch_ctypes_bindings.py#L38 ), but tell me that I can define some environment variable which specifies the library path.

This means any user of any program using the xnvme python bindings can ultimately provide the library path if the library loader fails to find the library.

jwdevantier, 2023-07-24:
PS - use `textwrap.dedent` to rewrite the code snippets in the auxiliary `patch_ctypes_bindings.py` file as multiline strings, while getting the right indentation.

safl, 2023-08-14:
Briefly, could 2) be described as:

For the xNVMe Python bindings; add the ability to override library-search, in library loading, by using path given via environment variable ``XNVME_LIBRARY_PATH``?