xnvme/xnvme #310: Segafault in kvs_io at strlen

agupta2303, 2023-07-25:
stack trace :  

Program received signal SIGSEGV, Segmentation fault.
__strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:74
74      ../sysdeps/x86_64/multiarch/strlen-avx2.S: No such file or directory.
(gdb) bt
#0  __strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:74
#1  0x000055555556fbed in **kvs_io** (cli=0x555555695680 <g_cli>) at ..**/tests/kvs.c:30**
#2  0x00005555555737e5 in xnvme_cli_run (cli=0x555555695680 <g_cli>, argc=9, argv=0x7fffffffe4c8, opts=1) at ../lib/xnvme_cli.c:1840
#3  0x000055555556ffc9 in main (argc=9, argv=0x7fffffffe4c8) at ../tests/kvs.c:116

This seems to due to incomplete assignment of kv_key and kv_val pointers.

Command executed was :   "./xnvme_tests_kvs kvs_io /dev/nvme0n1  --key "tango" --value "charlie" --be linux"

karlowich, 2023-07-25:
Hi @agupta2303, before I can help you I would like to know some things about your system.
Which OS/distro are you running?
Which version/branch of xNVMe are you using?

> `/sysdeps/x86_64/multiarch/strlen-avx2.S: No such file or directory.`

This looks like maybe you have a cross-compilation issue, are you compiling xNVMe on the system running it?

Also, if you recompile xNVMe with debug enabled, e.g., by running `make config-debug`, it will be easier to debug your issue.

safl, 2023-08-09:
Hi @agupta2303,
The fix that @karlowich is available on the ``next`` branch and will be available with ``v0.7.1``.
Did you have a chance to try it out for yourself?

safl, 2023-08-14:
Hi @agupta2303 , closing this i consider it fixed on ``next``. Feel free to re-open it.

PR #311 (2023-07-28): Fix KVS test
There were two oversights in the kvs test, these have been corrected.

Additionally, a cijoe test has been added using the functionality that previously failed.
This should resolve #310
closed 2023-08-14 by safl