xnvme/xnvme #313: write via xNVMe pioc command

agupta2303, 2023-07-27:
I tried xnvme pioc command. I had figure a lot from the test cases, can we please add a documentation around "pioc" and "padc", so others don't have to face similar issue.

Also can we add detail regarding, how do we give data-input through "pioc" command, if i have a data input file to write via "xnvme pioc" 

I tried the following 

**1. created a cmd-input file** 
`root@xnvme-mac1:~# nvmec create --opcode 0x1 --nsid 0x1 --cmd-output /tmp/write.nvmec
# DBG:xnvme_be.c:xnvme_be_factory-652: INFO: checking be: 'spdk'
# DBG:xnvme_be.c:xnvme_be_factory-683: INFO: obtained backend instance be: 'spdk'
# DBG:xnvme_be_spdk_dev.c:xnvme_be_spdk_state_init-761: FAILED: dev->opts.nsid: 0,
# DBG:xnvme_be_spdk_dev.c:xnvme_be_spdk_dev_open-890: FAILED: xnvme_be_spdk_state_init()
# DBG:xnvme_be.c:xnvme_be_factory-703: INFO: skipping backend due to err: -22
# DBG:xnvme_be.c:xnvme_be_factory-652: INFO: checking be: 'linux'
# DBG:xnvme_be.c:xnvme_be_factory-683: INFO: obtained backend instance be: 'linux'
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-50: INFO: open() : opts->oflags: 0xa, flags: 0x41, opts->create_mode: 0x180
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-77: INFO: open() : regular file
# DBG:xnvme_be.c:xnvme_be_dev_idfy-490: INFO: !xnvme_adm_idfy_ctrlr_csi(CSI_ZONED)
# DBG:xnvme_be.c:xnvme_be_dev_idfy-514: INFO: no positive response to idfy(ZNS)
# DBG:xnvme_be.c:xnvme_be_dev_idfy-548: INFO: looks like csi(FS)
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-193: INFO: open() : dev->state.poll_io: 0
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-194: INFO: open() : dev->state.poll_sq: 0
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-196: INFO: --- open() : OK ---
# DBG:xnvme_be.c:xnvme_be_factory-688: INFO: obtained device handle
root@xnvme-mac1:~#
`
output of nvmec show command :
`root@xnvme-mac1:~# nvmec show --cmd-input /tmp/write.nvmec
# DBG:xnvme_be.c:xnvme_be_factory-652: INFO: checking be: 'spdk'
# DBG:xnvme_be.c:xnvme_be_factory-683: INFO: obtained backend instance be: 'spdk'
# DBG:xnvme_be_spdk_dev.c:xnvme_be_spdk_state_init-761: FAILED: dev->opts.nsid: 0,
# DBG:xnvme_be_spdk_dev.c:xnvme_be_spdk_dev_open-890: FAILED: xnvme_be_spdk_state_init()
# DBG:xnvme_be.c:xnvme_be_factory-703: INFO: skipping backend due to err: -22
# DBG:xnvme_be.c:xnvme_be_factory-652: INFO: checking be: 'linux'
# DBG:xnvme_be.c:xnvme_be_factory-683: INFO: obtained backend instance be: 'linux'
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-50: INFO: open() : opts->oflags: 0x1, flags: 0x0, opts->create_mode: 0x0
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-77: INFO: open() : regular file
# DBG:xnvme_be.c:xnvme_be_dev_idfy-490: INFO: !xnvme_adm_idfy_ctrlr_csi(CSI_ZONED)
# DBG:xnvme_be.c:xnvme_be_dev_idfy-514: INFO: no positive response to idfy(ZNS)
# DBG:xnvme_be.c:xnvme_be_dev_idfy-548: INFO: looks like csi(FS)
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-193: INFO: open() : dev->state.poll_io: 0
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-194: INFO: open() : dev->state.poll_sq: 0
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-196: INFO: --- open() : OK ---
# DBG:xnvme_be.c:xnvme_be_factory-688: INFO: obtained device handle
xnvme_spec_cmd:
  - '00000000000000000000000000000001'
  - '00000000000000000000000000000001'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
  - '00000000000000000000000000000000'
`

**2. I run the following command** 
`root@xnvme-mac1:~# xnvme pioc /dev/nvme0n1 --cmd-input /tmp/write.nvmec --data-nbytes 4096
# DBG:xnvme_be.c:xnvme_be_factory-652: INFO: checking be: 'spdk'
# DBG:xnvme_be.c:xnvme_be_factory-683: INFO: obtained backend instance be: 'spdk'
[2023-07-27 19:34:47.816687] Starting SPDK v22.09 git sha1 aed4ece93 / DPDK 22.07.0 initialization...
[2023-07-27 19:34:47.816743] [ DPDK EAL parameters: [2023-07-27 19:34:47.816754] spdk [2023-07-27 19:34:47.816761] --no-shconf [2023-07-27 19:34:47.816766] -c 0x1 [2023-07-27 19:34:47.816771] --huge-unlink [2023-07-27 19:34:47.816778] --log-leve
l=lib.eal:6 [2023-07-27 19:34:47.816783] --log-level=lib.cryptodev:5 [2023-07-27 19:34:47.816788] --log-level=user1:6 [2023-07-27 19:34:47.816793] --iova-mode=pa [2023-07-27 19:34:47.816798] --base-virtaddr=0x200000000000 [2023-07-27 19:34:47.81
[... 29 lines cut ...]
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-193: INFO: open() : dev->state.poll_io: 0
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-194: INFO: open() : dev->state.poll_sq: 0
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-196: INFO: --- open() : OK ---
# DBG:xnvme_be.c:xnvme_be_factory-688: INFO: obtained device handle
# xnvme_cmd_pass(...)
# DBG:xnvme_be.c:xnvme_be_factory-652: INFO: checking be: 'spdk'
# DBG:xnvme_be.c:xnvme_be_factory-683: INFO: obtained backend instance be: 'spdk'
# DBG:xnvme_be_spdk_dev.c:xnvme_be_spdk_state_init-761: FAILED: dev->opts.nsid: 0,
# DBG:xnvme_be_spdk_dev.c:xnvme_be_spdk_dev_open-890: FAILED: xnvme_be_spdk_state_init()
# DBG:xnvme_be.c:xnvme_be_factory-703: INFO: skipping backend due to err: -22
# DBG:xnvme_be.c:xnvme_be_factory-652: INFO: checking be: 'linux'
# DBG:xnvme_be.c:xnvme_be_factory-683: INFO: obtained backend instance be: 'linux'
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-50: INFO: open() : opts->oflags: 0x1, flags: 0x0, opts->create_mode: 0x0
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-77: INFO: open() : regular file
# DBG:xnvme_be.c:xnvme_be_dev_idfy-490: INFO: !xnvme_adm_idfy_ctrlr_csi(CSI_ZONED)
# DBG:xnvme_be.c:xnvme_be_dev_idfy-514: INFO: no positive response to idfy(ZNS)
# DBG:xnvme_be.c:xnvme_be_dev_idfy-548: INFO: looks like csi(FS)
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-193: INFO: open() : dev->state.poll_io: 0
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-194: INFO: open() : dev->state.poll_sq: 0
# DBG:xnvme_be_linux_dev.c:xnvme_be_linux_dev_open-196: INFO: --- open() : OK ---
# DBG:xnvme_be.c:xnvme_be_factory-688: INFO: obtained device handle
# DBG:xnvme_be_linux_nvme.c:ioctl_wrap-138: INFO: ioctl(NVME_IOCTL_IO64_CMD), err(0), errno(0)
`

It seems to generate no error, but am confused regarding what is written on the device.

safl, 2023-07-28:
Hi @agupta2303 ,

Thanks for your input on this. With ``xnmve pioc`` and ``xnvme padc`` then you can "passthru" I/O and Admin commands, where you as the user, define the 64byte command construction. The intent is to use this for commands that are not defined in any command-set specifications. For those, then there are tools such as:

```
lblk write
kvs store
zoned append
```

For the command that you are constructing here (the write-command from the NVM Command-Set), then you could use ``lblk write --help`` instead. It has a better description of how the arguments work. To understand better what you are requesting. Would you like to send NVM commands or send arbitrary user-defined commands?

agupta2303, 2023-07-28:
Hi @safl,

Thanks for your insight. The intent is to use it as a user defined 64 byte command construction going ahead.
 But right now I was trying with the standard command set of read and write. I see in the  test tool "**cijoe/tests/tool/test_xnvme.py**" the pioc command getting used for standard command set "read".

In the below test case, can you help with the working of pioc, where is read output directed or how can we print that in stdout or any data output file.(similarly for opcode of 0x1(write))

> 205 def **test_pioc**(cijoe, device, be_opts, cli_args):
206     """Construct and send an I/O command (read)"""
207
208     opcode = "0x02"
209     cmd_path = "/tmp/cmd-out.nvmec"
210
211     err, state = cijoe.run(f"xnvme info {cli_args}")
212     assert not err
213
214     data_nbytes = 0
215     for line in state.output().split("\n"):
216         if "lba_nbytes" not in line:
217             continue
218         key, val = line.split(":")
219         data_nbytes = int(val.strip())
220         break
221
222     assert data_nbytes > 0
223
224     err, _ = cijoe.run(f"rm {cmd_path}")
225
226     err, _ = cijoe.run(
227         f"nvmec create"
228         f" --opcode {opcode}"
229         f" --nsid {device['nsid']}"
230         f" --cmd-output {cmd_path}"
231     )
232     assert not err
233
234     err, _ = cijoe.run(f"nvmec show --cmd-input {cmd_path}")
235     assert not err
236
237     err, _ = cijoe.run(
238         f"xnvme pioc {cli_args} --cmd-input {cmd_path} --data-nbytes {data_nbytes}"
239     )
240     assert not err

safl, 2023-07-31:
> Hi @safl,
>
> Thanks for your insight. The intent is to use it as a user defined 64 byte command construction going ahead. But right now I was trying with the standard command set of read and write. I [...]

You would invoke it with the ``--data-input`` argument like like so:
```
xnvme pioc /dev/nvme0n1 --cmd-input /tmp/write.nvmec --data-nbytes 4096 --data-input /tmp/your-input.bin
```
Same goes for ``xnvme padc ...``.

### Regarding documentation

Currently xNVMe has documentation for the cli via ``--help`` usage and ``man-pages``.
E.g. ``xnvme pioc --help`` will tell you:

```
xnvme pioc --help
Usage: xnvme pioc <uri> [<args>]

Pass a used-defined IO Command through

Positional arguments:

  uri                           ; Device URI e.g. '/dev/nvme0n1', '0000:01:00.1', '10.9.8.1.8888', '\\.\PhysicalDrive1'

Where <args> include:

  [ --nsid 0xNUM ]              ; Namespace Identifier for Command Construction
  --cmd-input FILE              ; Path to command input-file
  [ --data-input FILE ]         ; Path to data input-file
  [ --data-output FILE ]        ; Path to data output-file
  [ --data-nbytes NUM ]         ; Data size in bytes
  [ --meta-input FILE ]         ; Path to meta input-file
  [ --meta-output FILE ]        ; Path to meta output-file
  [ --meta-nbytes NUM ]         ; Meta size in bytes

With <args> for backend:

  [ --subnqn STRING ]           ; Subsystem NQN of the NVMe over Fabrics endpoint e.g. 'nqn.2022-06.io.xnvme:ctrlnode1'
  [ --hostnqn STRING ]          ; The host NQN to use when connecting to NVMe over Fabrics controllers
  [ --be STRING ]               ; xNVMe backend, e.g. 'linux', 'spdk', 'fbsd', 'macos', 'posix', 'windows'
  [ --dev-nsid 0xNUM ]          ; Namespace Identifier for Device Handle
  [ --admin STRING ]            ; xNVMe admin. command-interface, e.g. 'nvme', 'block'
  [ --mem STRING ]              ; xNVMe buffer/memory manager
  [ --direct NUM ]              ; Bypass layers
  [ --help ]                    ; Show usage / help

See 'xnvme --help' for other commands

xNVMe - Cross-platform NVMe utility -- ver: {major: 0, minor: 7, patch: 0}
```

There is also the man-page: ``man xnvme-pioc``

The documentation you find lacking is that because you are expecting some usage-examples in the online docs on https://xnvme.io?

### Regarding the CLI

So, four years ago or so,  I thought providing the command as a file e.g. the ``/tmp/write.nvmec`` was really clever. However, It has not seen much usage, rather an interface providing the command-dwords and some common flags seem to be favorable. So, for your use-case, would you prefer the current cli-arguments (and have them documented with an example) or another cli-interface of:

```
xnvme pioc --help
Usage: xnvme pioc <uri> [<args>]

Pass a used-defined IO Command through

Positional arguments:

  uri                           ; Device URI e.g. '/dev/nvme0n1', '0000:01:00.1', '10.9.8.1.8888', '\\.\PhysicalDrive1'

Where <args> include:

  [ --opcode 0xNUM ]            ; Command Opcode
  [ --nsid 0xNUM ]              ; Namespace Identifier for Command Construction
  [ --cdw0 0xNUM ]              ; Command-Dword 0 of the 64byte NVMe command
  [ --cdw1 0xNUM ]              ; Command-Dword 1 of the 64byte NVMe command
  [ --cdw2 0xNUM ]              ; Command-Dword 2 of the 64byte NVMe command
  [ --cdw3 0xNUM ]              ; Command-Dword 3 of the 64byte NVMe command
  [ --cdw4 0xNUM ]              ; Command-Dword 4 of the 64byte NVMe command
  [ --cdw5 0xNUM ]              ; Command-Dword 5 of the 64byte NVMe command
  [ --cdw6 0xNUM ]              ; Command-Dword 6 of the 64byte NVMe command
  [ --cdw7 0xNUM ]              ; Command-Dword 7 of the 64byte NVMe command
  [ --cdw8 0xNUM ]              ; Command-Dword 8 of the 64byte NVMe command
  [ --cdw9 0xNUM ]              ; Command-Dword 9 of the 64byte NVMe command
  [ --cdw10 0xNUM ]             ; Command-Dword 10 of the 64byte NVMe command
  [ --cdw11 0xNUM ]             ; Command-Dword 11 of the 64byte NVMe command
  [ --cdw12 0xNUM ]             ; Command-Dword 12 of the 64byte NVMe command
  [ --cdw13 0xNUM ]             ; Command-Dword 13 of the 64byte NVMe command
  [ --cdw14 0xNUM ]             ; Command-Dword 14 of the 64byte NVMe command
  [ --cdw15 0xNUM ]             ; Command-Dword 15 of the 64byte NVMe command
  [ --data-input FILE ]         ; Path to data input-file
  [ --data-output FILE ]        ; Path to data output-file
  [ --data-nbytes NUM ]         ; Data size in bytes
  [ --meta-input FILE ]         ; Path to meta input-file
  [ --meta-output FILE ]        ; Path to meta output-file
  [ --meta-nbytes NUM ]         ; Meta size in bytes

With <args> for backend:

  [ --subnqn STRING ]           ; Subsystem NQN of the NVMe over Fabrics endpoint e.g. 'nqn.2022-06.io.xnvme:ctrlnode1'
  [ --hostnqn STRING ]          ; The host NQN to use when connecting to NVMe over Fabrics controllers
  [ --be STRING ]               ; xNVMe backend, e.g. 'linux', 'spdk', 'fbsd', 'macos', 'posix', 'windows'
  [ --dev-nsid 0xNUM ]          ; Namespace Identifier for Device Handle
  [ --admin STRING ]            ; xNVMe admin. command-interface, e.g. 'nvme', 'block'
  [ --mem STRING ]              ; xNVMe buffer/memory manager
  [ --direct NUM ]              ; Bypass layers
  [ --help ]                    ; Show usage / help

See 'xnvme --help' for other commands

xNVMe - Cross-platform NVMe utility -- ver: {major: 0, minor: 7, patch: 0}
```
In other words, moving the cli-arguments from ``nvmec`` to ``xnvme pioc`` and ``xnvme padc`` such that an intermediate command-file is not needed. Let me know your thoughts on this.

safl, 2023-08-09:
Hi @agupta2303 ,

The behavior is now changed and documented. The new form is available on the ``next`` branch and will be available with the next release (``v0.7.1``). For documentation and examples of passthru via the command-line, then have a look at: https://xnvme.io/docs/next/tools/xnvme/pioc/index.html

safl, 2023-08-14:
Hi @agupta2303 , feel free to re-open this if you have feedback on the changes introduced in the cli and the description in the documentation.

closed 2023-08-14 by safl