xnvme/xnvme #319: Completion-handling via eventfd

safl, 2023-08-08:
API Changes for eventfd completion

- [ ] Experimental API to handle eventfd
  - [ ] Helper-function returning eventfd from queue ```int xnvme_queue_get_eventfd(struct xnvme_queue *queue);```
  - [ ] Some other heplers needed?

Example Application

- [ ] Cli-tool using event-driven I/O to read a single lba
  - In the same manner as the cli programs in ``examples/*.c``
  - Possibly similar to ``examples/xnvme_single_async.c``

Initial backend implementation in:

- [ ] The vfio-backend (libvfn)
- [ ] The io_uring backend (io_uring)
- [ ] The io_uring backend command (io_uring_cmd)

karlowich, 2025-02-11:
@safl Can we close this? or do we wait for tests using libblkio?

safl, 2025-02-11:
> [@safl](https://github.com/safl) Can we close this? or do we wait for tests using libblkio?

It is still open on libblkio https://gitlab.com/libblkio/libblkio/-/merge_requests/168
So, lets keep it open for now as well.