xnvme/xnvme #324: VFIO outputs an error on enumeration in CI

karlowich, 2023-08-11:
In the *Execute 'xnvme enum'* step on distros with libvfn enabled, the libvfn backend outputs the following debug message:
```
xnvme enum
D vfio_init_container (src/vfio/core.c:60): vfio/core: failed to open vfio device
D open_default_container (src/vfio/core.c:93): vfio/core: default container not initialized
xnvme_cli_enumeration:~
```

This should be silenced

karlowich, 2023-10-20:
With #366 the "noise" actually got worse, now libvfn outputs the following:
```
xnvme enum
D init_page_size (src/support/mem.c:42): pagesize is 4096 (shift 12)
xnvme_cli_enumeration:~
D measure_ticks_freq (src/support/ticks.c:37): measuring tick frequency
D init_ticks_freq (src/support/ticks.c:85: tick frequency is ~2290000000 Hz
D vfio_init_container (src/vfio/core.c:60): vfio/core: failed to open vfio device
D open_default_container (src/vfio/core.c:93): vfio/core: default container not initialized
D init_max_prps (src/nvme/rq.c:56): max prps is 513
```

birkelund, 2023-10-20:
You are using a debug build of libvfn. Build it using `--buildtype=release` instead.

karlowich, 2023-10-30:
Since #369 has been merged, I am closing this.

PR #369 (2023-10-25): fix(toolbox/pkgs): build libvfn as release
Currently we are building libvfn with debug enabled, this leads to unwanted debug messages getting printed in our CI. Silence these by building as release.

This closes #324
closed 2023-10-30 by karlowich