xnvme/xnvme #379: How can I get the list of supported command sets?

Neurone, 2023-11-09:
Hi, as a segue to #134, I'm trying to get the list of the supported command sets on my Samsung 990 Pro.

I tried the `idfy-cs` subcommand, but I get the following error. Any ideas?

```
❯ sudo xnvme idfy-cs /dev/nvme0n1
# ERR: 'xnvme_cmd_pass_admin()': {errno: -5, msg: 'Input/output error'}
xnvme_cmd_ctx: {cdw0: 0x0, sc: 0x2, sct: 0x0}
# ERR: 'idfy-cs': {errno: -5, msg: 'Input/output error'}

❯ sudo xnvme idfy-cs /dev/nvme0n1 --dev-nsid 0x1
# ERR: 'xnvme_cmd_pass_admin()': {errno: -5, msg: 'Input/output error'}
xnvme_cmd_ctx: {cdw0: 0x0, sc: 0x2, sct: 0x0}
# ERR: 'idfy-cs': {errno: -5, msg: 'Input/output error'}

❯ sudo xnvme idfy-cs /dev/nvme0
# ERR: 'xnvme_cmd_pass_admin()': {errno: 1, msg: 'Operation not permitted'}
xnvme_cmd_ctx: {cdw0: 0x0, sc: 0x3, sct: 0x3}
# ERR: 'idfy-cs': {errno: 1, msg: 'Operation not permitted'}
```

I'm using Ubuntu 20.04.3:

```
❯ uname -a
Linux dev 5.15.0-86-generic #96~20.04.1-Ubuntu SMP Thu Sep 21 13:23:37 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

And it seems I can use other commands without problems, like:

```
❯ sudo xnvme list
xnvme_cli_enumeration:
  - uri: /dev/nvme0n1

  - uri: /dev/ng0n1

❯ sudo xnvme info /dev/nvme0n1
xnvme_dev:
  xnvme_ident:
    uri: '/dev/nvme0n1'
    dtype: 0x2
    nsid: 0x1
    csi: 0x0
    subnqn: 'nqn.1994-11.com.samsung:nvme:990PRO:M.2:S6Z1NJ0W246524B     '
  xnvme_be:
    admin: {id: 'nvme'}
    sync: {id: 'nvme'}
    async: {id: 'emu'}
    attr: {name: 'linux'}
  xnvme_opts:
    be: 'linux'
    mem: 'posix'
    dev: 'FIX-ID-VS-MIXIN-NAME'
    admin: 'nvme'
    sync: 'nvme'
    async: 'emu'
    oflags: 0x4
  xnvme_geo:
    type: XNVME_GEO_CONVENTIONAL
    npugrp: 1
    npunit: 1
    nzone: 1
    nsect: 1953525168
    nbytes: 512
    nbytes_oob: 0
    tbytes: 1000204886016
    mdts_nbytes: 65024
    lba_nbytes: 512
    lba_extended: 0
    ssw: 9

❯ sudo xnvme idfy-ctrlr /dev/nvme0n1
xnvme_spec_idfy_ctrlr:
  vid: 0x144d
  ssvid: 0x144d
  sn: 'S6Z1NJ0W246524B     '
  mn: 'Samsung SSD 990 PRO 1TB                 '
  fr: '3B2QJXD7'
  rab: 2
  ieee: 002538
  cmic: 0
  mdts: 9
  cntlid: 0x1
  ver: 0x20000
  rtd3r: 0x30d40
  rtd3e: 0x989680
  oaes: 0x200
  ctratt: 0x10
  rrls: 0
  cntrltype: 0x1
  crdt1: 0
  crdt2: 0
  crdt3: 0
  nvmsr: 0
  vwci: 0
  mec: 0
  oacs: 0x17
  acl: 7
  aerl: 3
  frmw: 0x16
  lpa: { ns_smart: 1, celp: 1, edlp: 1, telemetry: 1, pel: 0, mel: 1, tel_da4: 0, val: 0x2f}
  elpe: 63
  npss: 4
  avscc: 0x1
  apsta: 0x1
  wctemp: 355
  cctemp: 358
  mtfa: 0
  hmpre: 0
  hmmin: 0
  tnvmcap: [1000204886016, 0]
  unvmcap: [0, 0]
  rpmbs: 0
  edstt: 35
  dsto: 0
  fwug: 0
  kas: 0
  hctma: 0x1
  mntmt: 318
  mxtmt: 356
  sanicap: 0x60000003
  hmminds: 0
  hmmaxd: 0
  nsetidmax: 0
  endgidmax: 1
  anatt: 0
  anacap: 0
  anagrpmax: 0
  nanagrpid: 0
  pels: 0
  domain_identifier: 0
  megcap: [0, 0]
  sqes: 0x66
  cqes: 0x44
  maxcmd: 256
  nn: 1
  oncs: 0x55
  fuses: 0
  fna: 0x7
  vwc: 0x7
  awun: 1023
  awupf: 0
  nvscc: 1
  acwu: 0
  cdfs: 0
  sgls: 0
  mnan: 0
  maxdna: [0, 0]
  maxcna: 0
  subnqn: 'nqn.1994-11.com.samsung:nvme:990PRO:M.2:S6Z1NJ0W246524B     '

karlowich, 2023-11-10:
`xnvme idfy-cs` sends an identify command with the CNS value `0x1C` (I/O Command Set data structure)  which according to the spec is optional. 
Most likely your command is failing because the drive does not support this CNS value.

Neurone, 2023-11-10:
Hi @karlowich, thanks for the kind answer.

Some side questions:
- Is there another way to discover what Command Sets a drive supports?
- Is the "{errno: -5, msg: 'Input/output error'}" error always a drive error? I think I've seen the same error in case I just use the wrong paramter in the CLI.
- Is there a way to activate something like a verbose/tracing mode, where I can see, for example, the `idfy-cs` command sending 0x1C and what is the actual answer from the drive?

karlowich, 2023-11-13:
Hi @Neurone,

> Is there another way to discover what Command Sets a drive supports?

As far as I know, there is no other way to tell, but maybe @safl knows something I don't.

> Is the "{errno: -5, msg: 'Input/output error'}" error always a drive error? I think I've seen the same error in case I just use the wrong paramter in the CLI.

You are correct, unfortunately xNVMe can return `EIO 5 Input/output error` for a variety of reasons this is something we are working on improving. If you find a place where we return `EIO` but could return a more descriptive/fitting error, you are welcome to create a pull request :)

 > Is there a way to activate something like a verbose/tracing mode, where I can see, for example, the idfy-cs command sending 0x1C and what is the actual answer from the drive?
 
 You can enable debugging by either running `make config-debug` or `meson setup builddir --buildtype=debug`, this will give you more insight into what is going on 'behind the scenes'.