xnvme/xnvme #401: Add SO-versioning

safl, 2023-11-29:
Packaging for Debian + Fedora / Rocky / RHEL, then so-versioning is a requirement.
Currently, the so-library is just plainly named ``libxnvme.so``. Thus, a strategy for so-versioning is needed, something that will complement the semantic versioning of the project itself.

karlowich, 2023-11-29:
@safl In meson we seem to have two options according to the  [docs](https://mesonbuild.com/Reference-manual_functions.html#shared_library).

Either `version`:
"On Linux and OS X, this is used to set the shared library version in the filename, such as `libfoo.so.1.1.0` and `libfoo.1.1.0.dylib`"

Or `soversion` :
"On Linux and Windows this is used to set the soversion (or equivalent) in the filename. For example, if soversion is 4, a Windows DLL will be called `foo-4.dll` and one of the aliases of the Linux shared library would be `libfoo.so.4`"

NOTE: if `soversion` is not set: "the first part of version is used instead. For example, if version is 3.6.0 and soversion is not defined, it is set to 3." 

In my opinion this behavior is undesired, since all of the past xNVMe releases would include `libxnvme.so.0` indicating that they are interchangeable - which they are certainly not.

safl, 2023-11-29:
Right, question is then which convention for SOVERSION to follow. To name few, then there is [libtool](https://www.gnu.org/software/libtool/manual/html_node/Updating-version-info.html) and FreeBSD-style symbol-versioning (https://people.freebsd.org/~deischen/symver/library_versioning.txt). However, unless it is too unconventional, then I would prefer to simply use the **SemVer** on the SOVERSION, however, I am not sure what kind of issues that might raise... from a quick look then others do so on Debian:

- https://sources.debian.org/src/lib2geom/1.3-2/debian/lib2geom1.3.0.install/
- https://sources.debian.org/src/adplug/2.3.3%2Bdfsg-2/debian/control/

However, there are plenty others that do things like:

- **SemVer** on the project, use `MAJOR` as `SOVERSION`, e.g. v2.3.4, `SOVERSION=2`
- **SemVer** on the project, a single integer for `SOVERSION`, bump it for **any** change to the symbols e.g. could have v2.3.4 with `SOVERSION=42`

I would like to understand the motivations and trade-offs behind these different conventions... however... the immediate motivation is to just go with using the **SymVer** on the SOVERSION...

safl, 2023-12-05:
Currently going with:

- SemVer on the project, use MAJOR as SOVERSION, e.g. v2.3.4, SOVERSION=2

This produces filenames ``libxnvme.so.0`` and ``libxnvme.so.0.7.3``, additionally then Debian requires a symbol-table e.g. ``libxnvme0.symbols``, this will contain symbol-versioning and provide more details on what is available and in which versions. I have taken a stab at producing, however, I do expect changes once the Debian package gets a sponsor review.