xnvme/xnvme #419: build error cross compiling for 32 bit target

cpackham-atlnz, 2024-01-23:
I'm trying to add xnmve as a package to our in-house build system for embedded targets. I'm hitting some static asserts when compiling for 32-bit targets (arm, x86 and powerpc). The 64-bit targets (arm64, mips64, powerpc64, x86_x64) all seem happy.

The actual list of errors is pretty long but they're all along the same lines here's an example
```
armv7,xnvme: arm-unknown-linux-gnueabi-gcc -Ilib/libxnvme.so.p -Ilib -I../lib -I. -I.. -Iinclude -I../include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=gnu11 -O3 -Wno-missing-braces -Wno-cast-function-type -Wno-strict-aliasing -include xnvme_config.h --sysroot=/usr/src/output/armv7/xnvme/staging -O2 -g2 -gz -mtune=cortex-a9 -march=armv7-a -mabi=aapcs-linux -fPIC -pthread -MD -MQ lib/libxnvme.so.p/xnvme_be_cbi_admin_shim.c.o -MF lib/libxnvme.so.p/xnvme_be_cbi_admin_shim.c.o.d -o lib/libxnvme.so.p/xnvme_be_cbi_admin_shim.c.o -c ../lib/xnvme_be_cbi_admin_shim.c
armv7,xnvme: In file included from ../include/libxnvme.h:22,
armv7,xnvme:                  from ../include/xnvme_be.h:10,
armv7,xnvme:                  from ../lib/xnvme_be_cbi_admin_shim.c:8:
armv7,xnvme: ../include/libxnvme_util.h:16:40: error: static assertion failed: "Incorrect size"
armv7,xnvme:    16 | #define XNVME_STATIC_ASSERT(cond, msg) static_assert(cond, msg);
armv7,xnvme:       |                                        ^~~~~~~~~~~~~
armv7,xnvme: ../include/libxnvme_cmd.h:35:1: note: in expansion of macro 'XNVME_STATIC_ASSERT'
armv7,xnvme:    35 | XNVME_STATIC_ASSERT(sizeof(struct xnvme_cmd_ctx) == 128, "Incorrect size")
armv7,xnvme:       | ^~~~~~~~~~~~~~~~~~~
```
The full log is attached [build.log](https://github.com/OpenMPDK/xNVMe/files/14030467/build.log)

cpackham-atlnz, 2024-01-23:
Looking at the first failure there are a number of things in `struct xnvme_cmd_ctx` that are pointers so that would easily explain why the assert fails for 32-bit platforms.

safl, 2024-01-24:
Right, xNVMe is currently only tested on 64bit platforms. Support for 32bit would be great, one approach would be to expand the current build-testing that is done in the github-actions, doing cross-compilation for 32bit, and then going through it all.