xnvme/xnvme #450: Comparing the capabilities of `bdev_nvme.c` and `bdev_xnvme.c` with the NVMe specifications

NaddiNadja, 2024-04-03:
| specs | bdev_nvme.c | bdev_xnvme.c
-- | -- | -- | --
Flush | x | x |  
Write | x | x | x
Read | x | x | x
Write Uncorrectable | x |   |  
Compare | x | x |  
Write Zeroes | x | x | x
Dataset Management | x | x (unmap) |  
Verify | x |   |  
Copy | x | x |  
Compare and Write |   | x |  
Reset |   | x |  
Zone Append | x | x |  
Zone Management Receive | x | x (Get Zone Info) |  
Zone Management Send | x | x (Zone Management) |  
NVMe (passthru) |   | x |  
Abort |   | x |

jimharris, 2024-04-10:
Hi @NaddiNadja,

@karlowich asked me to take a look at your issue here and provide some feedback. I think this table is a great way to communicate the current gaps on bdev xnvme and will help prioritize those gaps.

I think a 4th column would be helpful here, for what xnvme supports. Then we can differentiate between where support is only needed in bdev_xnvme module, vs. what needs support in xnvme itself.

Maybe also a 5th column on relative priorities for the gaps?

For "compare and write", the support is there in the spec, look for "fused commands".  This is supporting two separate commands ("compare" and "write") where the controller executes the first command and only if it succeeds does it execute the second command. (Note: I think this is low priority for xnvme. The SPDK bdev layer can emulate the fused command support if the bdev doesn't support it.)

"Reset" is also in the spec, but this is not a command. It might be more described as a process. See section 3.7 in the 2.0c base spec.

I would also add rows for "hot remove" and "protection information".

"Hot remove" isn't part of NVMe spec, this is more of a system level requirement. bdev_nvme supports it, but I'm not sure if xnvme provides mechanism to notify user (i.e. bdev_xnvme) that a block device has been removed.

"Protection Information" is in the spec, it is not a command though, but rather a way to verify the correctness of read and write commands.

-Jim

NaddiNadja, 2024-04-16:
Hi @jimharris 

I have made the changes you requested. Karl mentioned that it would make sense to just remove the "specs" column, as it didn't make much sense to have, if it has x'es in every row. :)

The priority column is just empty for now. Perhaps you can help fill it out?
Let me know if you have other suggestions. 

- Nadja

jimharris, 2024-04-16:
Hi @NaddiNadja,

I don't seem to have edit privileges on issues. Maybe @karlowich can help me with that.

Let's say 1 is highest priority, 3 is lowest priority. Also S, M, L (small/medium/large) for a new "complexity" column.

Flush 3 S
Compare 3 S
Dataset management 1 S
Copy 2 S
Compare and write 3 L (I'm not sure this is possible with io_nvme_cmd backend)
Reset 1 L
Zone append/mgmt 2 M
NVMe Passthru 1 M
Abort 3 L (I'm not sure this is possible with io_nvme_cmd backend)
Hot Remove 1 L
PI 1 M

NaddiNadja, 2024-04-17:
Hi again!

Thank you for this. I have inserted the priorities in the table and added the complexity column.

NaddiNadja, 2024-11-05:
I'm gonna close this issue, as this table has been added to the documentation in PR #520 
https://xnvme.io/tutorial/bdev_xnvme/index.html

closed 2024-11-05 by NaddiNadja