xnvme/xnvme #546: Parser for XNVME_CLI_OPT_VTYPE_NUM parses hex value as 0

Baekalfen, 2024-10-25:
The parser tries to convert the string `0x4096` as a base-10 integer, and returns `0`. The order of the arguments are insignificant, and it will throw away the `x4096` part. The expected behavior would be to either parse the hex value correctly, or return an error.

```
./builddir/tools/lblk write [URI] ... --slba 0x0 --nlb 0x4096
# Writing nsid: 0x1, slba: 0x0000000000000000, nlb: 0
# Alloc/fill dbuf, dbuf_nbytes: 512
# Sending the command...
```

```
./builddir/tools/lblk write [URI] ... --slba 0x0 --nlb 4096
# Writing nsid: 0x1, slba: 0x0000000000000000, nlb: 4096
# Alloc/fill dbuf, dbuf_nbytes: 2097664
# Sending the command...
```

safl, 2024-10-25:
This is intended behavior, see usage of ``lblk`` below, ``nlb`` has has: ``--nlb NUM``  not ``--nlb 0xNUM``.

```
Usage: lblk write <uri> [<args>]

Writes data and optionally metadata

Positional arguments:

  uri                           ; Device URI e.g. '/dev/nvme0n1', '0000:01:00.1', '10.9.8.1.8888', '\\.\PhysicalDrive1'

Where <args> include:

  --slba 0xNUM                  ; Start Logical Block Address
  --nlb NUM                     ; Number of LBAs (NOTE: zero-based value)
  [ --nsid 0xNUM ]              ; Namespace Identifier for Command Construction
  [ --data-input FILE ]         ; Path to data input-file
  [ --meta-input FILE ]         ; Path to meta input-file

With <args> for backend:

  [ --subnqn STRING ]           ; Subsystem NQN of the NVMe over Fabrics endpoint e.g. 'nqn.2022-06.io.xnvme:ctrlnode1'
  [ --hostnqn STRING ]          ; The host NQN to use when connecting to NVMe over Fabrics controllers
  [ --be STRING ]               ; xNVMe backend, e.g. 'linux', 'spdk', 'fbsd', 'macos', 'posix', 'windows'
  [ --dev-nsid 0xNUM ]          ; Namespace Identifier for Device Handle
  [ --admin STRING ]            ; xNVMe admin. command-interface, e.g. 'nvme', 'block'
  [ --mem STRING ]              ; xNVMe buffer/memory manager
  [ --direct NUM ]              ; Bypass layers
  [ --sync STRING ]             ; xNVMe sync. command-interface, e.g. 'nvme', 'block'
  [ --help ]                    ; Show usage / help
```

Baekalfen, 2024-10-25:
I can see it's supposed to parse a hexadecimal and decimal number, but surely it's not intended to silently ignore these?

```lblk write 2GB --dev-nsid 1 --be ramdisk --admin ramdisk --slba 0x1HHHHHHHHHH --nlb 0KKKKKKK
# Writing nsid: 0x1, slba: 0x0000000000000001, nlb: 0
# Alloc/fill dbuf, dbuf_nbytes: 512
# Sending the command...
```

```lblk write 2GB --dev-nsid 1 --be ramdisk --admin ramdisk --slba 0x1HHHHHHHHHH --nlb 0KKKKKKK
# Writing nsid: 0x1, slba: 0x0000000000000001, nlb: 0
# Alloc/fill dbuf, dbuf_nbytes: 512
# Sending the command...
```

```
lblk write 2GB --dev-nsid 1 --be ramdisk --admin ramdisk --slba 0x1HHHHHHHHHH --nlb 0--async=NOT_A_BACKEND
# Writing nsid: 0x1, slba: 0x0000000000000001, nlb: 0
# Alloc/fill dbuf, dbuf_nbytes: 512
# Sending the command...
```

safl, 2024-10-28:
@Baekalfen To clarify, the parser is not "wrong". However, I agree that having input validation with descriptive error messages would be helpful. For instance, it could have saved a lot of confusion in this CLI usage. It would be great to add this feature.

karlowich, 2024-11-11:
Closing this since #547 has been merged

closed 2024-11-11 by karlowich