xnvme/xnvme #557: PCI Device enumeration

safl, 2024-11-26:
Currently, only the SPDK/NVMe backend supports PCIe device enumeration. The enumeration initializes the NVMe subsystem and attaches devices, but this occurs only once. Subsequent calls to xnvme_enumerate()—whether directly or indirectly (e.g., opening a device and then enumerating)—will only return devices that have not yet been "attached." This behavior is by design in SPDK and is not intended to be changed via patches to SPDK.

However, this approach is suboptimal from the user perspective in xNVMe. To improve this, enumeration should instead:

* Use PCI enumeration via the operating system,
  * For example, the equivalent of running ``lspci | grep "Non-Volatile memory controller"`` on Linux.

This method avoids requiring NVMe driver initialization to enumerate controllers. Instead, a similar approach could be implemented as part of the enumeration process, allowing the backend to handle device-open logic for each PCI address. This is feasible with SPDK as well, as it eliminates the need to rely on enumeration for attachment.

**Note**: This is a well-known issue recently encountered during testing. When using the xNVMe API via Python, multiple calls to open() or xnvme_enumerate() within the same Python process or library load can lead to problems. Specifically, a call to xnvme_enumerate() may fail to return a device because it was already attached by a previous test within the same process.

jadoc, 2024-12-29:
I don't think PCI level enumeration is needed. XNVMe already keeps a refcounted cache of all attached controllers. The enumeration logic just needs to be changed to iterate over both the cached already attached controllers AND the controllers returned by `spdk_nvme_probe()`.

I think attempting to implement custom parallel logic for identifying SPDK compatible PCIe NVMe devices would be error prone at best. Furthermore, your solution would not be future proofed against potential probe-able transports other than PCIe.

jadoc, 2024-12-30:
Or is there a concern that `spdk_nvme_probe()` won't work if the device is attached to a different SPDK process?

jadoc, 2025-03-15:
I think there are two separate enumeration issues with SPDK:

1. SDPK won't enumerate attached devices.
    * This is apparently a design choice in SPDK.
    * This can be worked around by using the existing refcount tracking to remember which devices are attached and manually include them in the enumerate.
2. Devices can't be re-enumerated within 2 seconds of the previous enumeration.
    * SPDK just enumerates controllers, while XNMe enumerates namespaces. This means that XNVMe has to attach each controller to enumerate its namespace then immediately detach it again.
    * When SPDK becomes aware of a new PCIe device, it waits 2 seconds before including it in the enumeration. This is to work around race conditions in kernel device initialization.
    * When a device is detached, a common next step is for it to be hot-swapped. If the device has been hot-swapped, SPDK needs to wait 2 seconds for kernel device initialization. If SPDK didn't do this, it would fail hot-swap stress tests.
    * Unfortunately, there is no good way for SPDK to tell if a device has actually been hot-swapped or not, so it must assume all detached devices were hot-swapped. This means that at least 2 seconds must pass before detached devices can be enumerated again.
    * A workaround for this in XNVMe could be to enforce that at least 2 seconds have passed since the last enumeration. XNMe could track the last enumeration time, and if it's been less than 2 seconds since the last enumeration, sleep until 2 seconds have been reached. This isn't ideal, but enumeration probably isn't on the critical path for performance in real applications.
    * Alternatively, SPDK could be modified to support the concept of a fast detach. The user would be able to explicitly indicate that they do not intend to swap the hardware after the detach, allowing immediate re-enumeration. I have no idea if the SPDK devs would be amenable to this.